fritzing/fritzing-app #3786: Change size of drillholes on board layout

atomroflman, 2021-02-18:
Hey Fritzing Team,

I am wondering, why it isn't possible to change the size of the drill holes for a component without creating a new one?
Couldn't the drill holes be added to the properties of a component?

KjellMorgenstern, 2021-02-18:
This would become available with implementation of #3746

mMerlin, 2021-03-03:
Actually, #3746 as written would allow for the inclusion of optional, and potentially multiple variations of drill holes in a part, but they would have to be explicitly included in each part. What is really wanted here, is to have access to the "advanced settings" properties available for the pin header and some other parts. I would extend that to have additional features, like turning all or specific pins from the standard circle into square or oblong pads. Fritzing already uses the position, radius, stroke-width from the svg file to define the properties. Conceptually it would seem practical to override those using (optional) information supplied through Inspector. Per part, per pin, or per group of pins (for parts with multiple headers).