fritzing/fritzing-app #3788: Fritzing 0.9.6 crashing on boot, Mac OS 10.13.6

dgeers, 2021-02-25:
I just paid for and installed Fritzing 0.9.6, because I love this software.  However when I boot it on my Macbook Pro (circa 2015) running High Sierra (10.13.6), it crashes immediately.

When I click the icon, the app appears and loads the parts. Then I see the main interface appear, but while it says "loading new sketch" and the main interface appears in the background, the entire app crashes.

Due to other apps I run, I'm not interested to move to the newer Mac OS.  I'd like to know if this can be fixed, or can I return to the previous version of Fritzing until I move to a newer OS?

Thanks much,

Doug

KjellMorgenstern, 2021-02-25:
The Fritzing 0.9.6 for Catalina and Big Sur was build on Catalina, and can not work on earlier versions.
Officially, High Sierra is not supported anymore.

However, I made a build that works on High Sierra and also Mojave. It is build on a fresh install of High Sierra, with latest security fixes (2020-006) and Qt 5.15.2 build from sources. I have uploaded it to fritzing.org, but naturally it is not as much tested as the other builds.

dgeers, 2021-02-25:
> CAUTION: This email is from outside BC, so examine it closely before opening attachments or clicking on links

OK.  Many thanks for your quick reply and action.

You are doing great work!  I appreciate this tool, as an artist and as an educator.

Mit freundlichen Grüssen,

Doug

closed 2021-02-25 by KjellMorgenstern