fritzing/fritzing-app #3792: It's hard to select objects in PCB view, not selecting the object the cursor is pointing at.

zn59, 2021-03-08:
This happened once and I had to reinstall the application. Now it happened again!  Must I reinstall twice a day?

I'm running on a MacBook Pro 2014 with latest Mac OS version and latest version of Fritzing, 0.9.6.

Best Regards
Zoltan

zn59, 2021-03-09:
Today I have to reinstall again but it doesn't help this time. Am the only one with this problem?

KjellMorgenstern, 2021-03-09:
Could you describe the problem in more detail? How does "hard to select" look like? And what gives the idea that reinstalling would solve the problem?

zn59, 2021-03-10:
Hi Kjell,

With hard to select I mean that it will not select the object I’m pointing at but a nearby object. To do something with a smaller object nearby a bigger is impossible and many times it grabs the PCB and moves it around. When it behaves like this reinstalling helped and worked a whole day, but next day when I start the application and open the project after a few actions in the PCB view the problem returns. Yesterday the reinstalling didn’t helped.

I don’t know if the problems can be related to downloading and importing of some parts, I think it was then the problems began. Reinstalling did not help, until I throw the application to bin before I began the installation. Then it worked again until next day… Can the problem relates to the part I use, the diode 1N4148 which is not in core parts?
I have added some parts in PCB view that is not in breadboard and schematic view, does it matter?

Best Regards

Zoltan Németh, Sweden

vanepp, 2021-03-10:
A couple of things that may help. To avoid the pcb moving, you can lock the part by selecting the part (the pcb background in this case) then right clicking and clicking lock part:

[image]

with this done the pcb won't move even if it gets selected until you unlock it again. For parts there are two issues, if the part has the silkscreen layer below the copper layers, silkscreen selects first, so yes some parts will make selection more difficult. You can sometimes (I find this often doesn't work though) select the part and then change its layer. In theory if you move a part to the back and move the part you want to the front, the front part will select first. To do that right click on the part and select raise and lower:

[image]

In practice this only seems to work sometimes and you need to keep trying until you get what you want which is frustrating.

zn59, 2021-03-11:
Hello,
Thank’s for the tips!  I’ll try. What I dont understand is why everyting works one day and next day not, and why I must click a bit away (usually below) the part I want to be selected. 

Best regards
Zoltan

zn59, 2021-03-11:
Hi…

Please test with this file. I have the same trouble in every view, breadboard, schematic… it’s impossible to select the right part and move it…

Best regards
Zoltan

 <https://user-images.githubusercontent.com/16514277/110704599-0c8e2e00-81aa-11eb-9f47-08900b83efd7.jpg>

 <https://user-images.githubusercontent.com/16514277/110704383-bde09400-81a9-11eb-973d-2b151d3f44e7.jpg>

zn59, 2021-03-11:
Hi again…

Just a few observations… started working on a new PCB, just a few VIA’s and capacitors and connections. No problem.

Picked up a just as simple PCB I made a few days ago… the problems are there immediately while switching back to the new PCB everything works as expected.

Maybe the same will happen with the new PCB when I open it tomorrow, who knows, but right now I have 2 windows open with the new and old PCB, containing identical parts and working with the new one is OK but not with the old one. Is'nt that odd?

Best Regards

Zoltan

 <https://user-images.githubusercontent.com/16514277/110704599-0c8e2e00-81aa-11eb-9f47-08900b83efd7.jpg>

 <https://user-images.githubusercontent.com/16514277/110704383-bde09400-81a9-11eb-973d-2b151d3f44e7.jpg>

zn59, 2021-03-11:
Hello…

Yet another observation. After saving my simple project, and quitting Fritzing (closing all windows) and then started it again and picked up the new PCB, then we are there again!  I must point somewhat below and to the right of the part to select it but it’s not always doable with other parts nearby. So something gets wrong after reopening a file!

Best Regards
Zoltan

 <https://user-images.githubusercontent.com/16514277/110704599-0c8e2e00-81aa-11eb-9f47-08900b83efd7.jpg>

 <https://user-images.githubusercontent.com/16514277/110704383-bde09400-81a9-11eb-973d-2b151d3f44e7.jpg>

failiz, 2021-03-12:
Hi,
Thanks for reporting the error. We need some extra information in order to fix it:
(1) Upload your file, and even better, (2) could you give us precise steps to reproduce your error?

The steps to reproduce it  could be something like:
1. Create a note
2. Click and begin typing
3. Try to delete note with delete key even after clicking out
etc.

zn59, 2021-03-12:
Hi,

I wish I could give You the precise steps but it’s hard….  

I can work on a new project with no problems and save it. Next time I pick it up to continue the work then it’s hard the select the object you want and do what you want. You have to point a bit right and below the object to get it highlighted and you can move it but can’t do a new connection, instead the object will be moved.

Yesterday a created a very simple PCB with some VIA’s, 2 capacitors and connections. No problems at all to move around the objects and changing the connections. Then I saved it and closed down the application completely. Started the application again, opened the file created previously and there we are…

Today I picked up my main project with lots of parts and I was very surprised that everything worked as supposed. Did some changes and saved it. Closed down the application, started it and opened the file and the buggy behaviour is back.  

I began working with this application just a few days ago and imported some parts and when it started to behave like this I thought that it has to do with my imported parts and reinstalled the application and after that it was OK until next day when I opened the file again. After reinstalling it worked again… But reinstalling did not help last time so I made the BUG report. And as I mentioned, I was surprised today when I could continue the work on my main project today, but after saving and opening again the error is back!  Seems very random to me!

Sending you a simple example that don’t work for me after saving and reopening. (I wonder why fritzing put some of the new sketches in the downloads folder….)
As I mentioned in the BUG report I’m running latest Mac OS and maybe only Mac users are affected this problem?

Best Regards
Zoltan

Create a note
Click and begin typing

failiz, 2021-03-12:
I don't see any attachments. You have to click on view in GitHub. And add the attachment in GitHub.

failiz, 2021-03-12:
And could you also check if there is any difference between opening the file using this two methods:
1. open Fritzing and the you go to file->open and then try to reproduce the bug in the file that you opened.
2. open Fritzing and a file at the same time (by double clicking a file or by passing the path of the file as argument in the command line).

vanepp, 2021-03-12:
It would be most useful to get a copy of the sketch.fzz file for a failing sketch. That will tell us if it happens on other machines or is something specific to your machine. On Windows (where I am) I need to rename the test.fzz (sketch) file to test.fzz.zip because Github won't upload .fzz files (as far as I know anyeay!) then do a click and drag of the file on my machine in to the github reply window. That uploads a copy of the file to github and it should appear in the reply window. Another thing it would be useful to try is to clear the user directories. They do not get cleared on a reinstall of the application (to preserver your modified parts across upgrades) but sometimes get corrupted. Here are instructions for clearing the directories:

"There are two user directories (with your parts and the parts database) which don't get touched during an install (to not affect your sketches during upgrades). On Windows they are in

c:\users\username\AppData\Fritzing\roaming\Fritzing (which is a hidden directory so you need to enable hidden directories in explorer) and

c:\Users\username\My Documents\Fritzing (where username is your windows id)

If you don't have any parts or sketches you want to keep you can just delete those two directories and Fritzing will recreate them, or you can move them aside by renaming them if you want to keep something in them.

linux

~/Documents/Fritzing/parts
~/.config/Fritzing 

Mac

/Users/username/Documents/Fritzing/parts
~/.config/Fritzing"

In your case you will want to move aside (for instance by renaming them) 

/Users/username/Documents/Fritzing/parts to /Users/username/Documents/Fritzing/parts.orig

and 

~/.config/Fritzing to ~/.config/Fritzing.orig

When Fritzing finds the directories are not present it will recreate new copies of them (but without any custom parts you have added to your mine parts bin.) If that clears the problem you can copy the original files back to the original file names and Fritzing will use them on the next start up. , You can then  export the parts in the mine parts bin by right clicking on the part and selecting export part, clear the user directories again and use File->load to reload the parts in to the mine parts bin. This is the first time I have heard of an error like this, so this may not help, but it has fixed a variety of odd faults in the past. The usual way they get corrupted is by interrupting Fritzing the first time it is started when it is updating the parts repository from github and appears to hang. If you interrupt it before the update completes the user directories usually get corrupted and need to be cleared.

zn59, 2021-03-12:
Hi,
As I wrote to fritzing-app, I finally noticed that the problem occurs in full screen mode!
I can avoide that mode for now and hope a fix will come in some future release.
Thank you all for the help and tips and have a nice weekend!

Best regards
Zoltan

vanepp, 2021-03-13:
"I finally noticed that the problem occurs in full screen mode!
I can avoide that mode for now and hope a fix will come in some future release."

In order to make a fix we need to be able to recreate the problem. So if you could upload a sketch (the .fzz file) that breaks in full screen mode (I'm hoping that how to get a Mac in to full screen mode is obvious to a Mac user here!) that would help get the problem fixed. 

Peter

zn59, 2021-03-13:
Hi…

Sending you this simple sketch. Now when testing again I learned that if the frizzing application is in full screen mode and I open any sketch in that mode it selects objects at a bit left and above from where I’m pointing. In the screen shot I’m pointing at the middle of the connection at the right of the VIA.

But if I leave full screen mode it works again as it should and even when I switch back to full screen mode!  So the issue is when opening the sketch in full screen mode…

Best regards

Zoltan

KjellMorgenstern, 2021-03-13:
I tried to reproduce problems on macOS BigSur, switching to fullscreen, but interaction had not
more than the usual problems. It is usable, with some known glitches, but it did not change compared window mode.
Maybe try to record a screen session together with the mouse clicks , so we can see the interaction:

Two tools which are able to also document keys and clicks are
[screencastify](https://www.screencastify.com/) and 
[obs](https://obsproject.com/)

zn59, 2021-03-13:
Hi Kjell,

If you open the sketch when in full screen mode then you should notice the problem. When you leave full screen mode the problem disappears even if you switch back. If you first open a sketch not in full screen mode you will never notice the problem...

Best regards
Zoltan

KjellMorgenstern, 2021-03-13:
@zn59 I think the file upload via the email reply did not work. Can you attach it with the browser frontend?

zn59, 2021-03-13:
Hi Kjell,

Here it comes, I could pick any of my files. As I explained the problem occurs ONLY if I open a sketch when the fritzing app is in full screen mode on my Mac and disappears when I leave full screen mode, even if I return to it afterwards...
If I open a sketch when NOT in full screen mode I never experience the problem, even if I switch to full screen mode.
I think when the application opens a sketch it never checks the screen mode... but when changing window to full screen mode it is notified about the change... ?

Best regards
Zoltan

[Sensor board.fzz.zip](https://github.com/fritzing/fritzing-app/files/6134737/Sensor.board.fzz.zip)

vanepp, 2021-03-13:
The sketch made it up (it needs to be unzipped to get the .fzz.) On windows it appears to work correctly so hopefully the issue will appear on a Mac in full screen mode (but I don't have  a Mac!)

zn59, 2021-03-14:
I could not drop the .fzz file directly, github did not allow it...
Maybe a developer with a Mac can test this:
1. Set the fritzing window to full screen mode.
2. Then, and first then (!) open any sketch. Then you will notice what I’m complaining about.

Now when I know how to handle it, the issue can be closed and I hope the code can be in some update in near future. The application is fantastic and my only wish is that parts could be modified more easily. Would like to delete or add a pin to a 5 pin DIN-connector.

KjellMorgenstern, 2024-12-08:
I can't reproduce this with current Fritzing. It likely was solved e.g by issue #3177 , see there for many details. I could also have been resolved by updated Qt and macOS.
Specifically, I could not see any offset when opening a sketch from a Fritzing running in full screen.

closed 2024-12-08 by KjellMorgenstern