fritzing/fritzing-app #3796: Building on Mageia Linux

Morganlej, 2021-03-15:
I am an occasional QA at Mageia, and an occasional Fritzing user.
Our packager have problems packaging Fritzing.
Specifically problems with creating parts.db
https://bugs.mageia.org/show_bug.cgi?id=18689#c32

Any hint?

Maybe https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes can be enhanced.

Thank you for this fun and educating program!
Now lets get it out to more users :)

david-geiger, 2021-03-18:
Hi,

I'm the mageia packager for fritzing package, some help is needed as I can't create/build the parts database (parts.db file) which is needed. I get the following error during build:

> ./Fritzing -platform minimal -f ./parts -db ./parts/parts.db
/home/iurt/rpmbuild/tmp/rpm-tmp.FO9JLJ: line 43: 23315 Aborted                 (core dumped) ./Fritzing -platform -f ./parts -db ./parts/parts.db
error: Bad exit status from /home/iurt/rpmbuild/tmp/rpm-tmp.FO9JLJ (%build)

And this one on my own system:

> ./Fritzing -platform minimal -f ./parts -db ./parts/parts.db
This plugin does not support propagateSizeHints()

Regards,
David

closed 2021-05-03 by KjellMorgenstern

Morganlej, 2021-05-03:
Reason for closure?

Should we simply skip building it and instead tell users to download the parts library to a folder and tell fritzing to use that?

( Like described at end of https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes )

KjellMorgenstern, 2021-05-03:
Sorry, i thought you have already solved it, but maybe i mixed something up. 
To ensure a wide support on Linux systems, we are currently testing an AppImage.

david-geiger, 2021-05-03:
Please re-open this bug as long as it is not fixed!

Morganlej, 2021-05-03:
Yes please reopen.

Meanwhile, i could help test if the Appimage works on Mageia 8-64 bit, if available.
Same, if you try to make a Flatpak.
(btw my user experience is that Flatpak programs in general have less problems than Appimages, but that may be just bad luck)

Morganlej, 2021-05-23:
Kjell?

Morganlej, 2021-08-03:
Ping...

Morganlej, 2022-03-29:
Anything?

Morganlej, 2023-03-18:
Sorry our issue tracker had not been updated.
Fritzing 0.9.10 is built and works in upcoming Mageia 9 :)

As a backup/support, I bought 0.9.10 and verified that the Appimage works on Mageia 8 :)