fritzing/fritzing-app #3798: Printing issue

zn59, 2021-03-19:
## Current Behaviour
I don't manage to print the whole PCB, a small bit on the left side is always missing. Tried both with or without margins but a bit of left side is always missing.

[image]

**Build:**
Version 0.9.6 
(bCD-175-0-8a1e0682 2021-02-21) Cocoa [Qt 5.15.2]

**Operating System:**
<OS X BigSur>

**Steps to reproduce:**
- Printing with different settings

## Expected Behaviour
Would like to se the whole PCB on paper.

KjellMorgenstern, 2021-03-19:
"Printing with different settings"

I am sorry, what should i do to reproduce this? Prints come out in original resolution, a 5cm wire will be 5cm on paper

zn59, 2021-03-20:
Hi Kjell,
The resolution is fine! What I’m complaining about is that i bit of the side is always  missing.
The two holes visible at the right side are there on the left side too...
I have tried with different settings, with or whithout margins but some bit to the left and top are not printed.

KjellMorgenstern, 2021-03-20:
[image]

KjellMorgenstern, 2021-03-20:
Can not reproduce

closed 2021-03-20 by KjellMorgenstern

zn59, 2021-03-21:
Windows or Mac? Which settings?