fritzing/fritzing-app #3800: What you se is NOT what you get when printing

zn59, 2021-03-23:
## Current Behaviour
As I explained in an earlier case (but case was closed, no errors found) a bit of the PCB is missing when printed out.

[ZN4 sensor board 1.0.pdf](https://github.com/fritzing/fritzing-app/files/6187374/ZN4.sensor.board.1.0.pdf)

The PDF document looks fine:

[image]

 BUT when printed out it looks like this:

[image]

**Build:**
Version 0.9.6 
(bCD-175-0-8a1e0682 2021-02-21) Cocoa [Qt 5.15.2]

**Operating System:**
<Mac OS X Big Sur>

**Steps to reproduce:**
1. Printing from fritzing.
2. Save as a PDF and then open the PDF and print... same result as in step 1.
 

## Expected Behaviour
Normally the printing preview shows exactly how it will look like on paper but not with fritzing or PDF created by fritzing.

Please try to print the attached PDF file, do you get the same result as I do?

KjellMorgenstern, 2021-03-23:
[image]

The provided pdf renders fine in macOS BigSur Preview, but shows artifacts after selecting / unselecting it by marking a rectangle with the mouse. The similar artifacts are also visible when printing out.
The same effect can be produced when exporting a pdf from Examples -> Processing -> Arduino

KjellMorgenstern, 2021-03-23:
[image]

zn59, 2021-03-24:
So where are we now? Did you print my PDF?

KjellMorgenstern, 2021-03-24:
Similar artifacts are visible when printing out. I could not see what you observed, but the print was messed up as shown above.

zn59, 2021-03-24:
In my case I can't see those artefacts, my observation/complain is that a bit of the left side is missing on the paper. The PDF document looks OK, as in my provided screenshot but on paper a bit of the left side is missing.

kittentronic, 2021-03-24:
If I print the PDF (from MacOS Catalina Preview on a Canon iP7250) the left side is there but the top edge is missing, and there are black blocks obscuring a large proportion of the circuit. Similarly if I open one of the Fritzing examples and export it to PDF then print it (in this case it's centred on the paper, but lots of black artefacts). In both cases the PDF contents and the preview in the print dialog look fine on screen.

Is the page margin issue a separate one from the artefacts, perhaps? Can you share the .fzz file?

KjellMorgenstern, 2021-03-24:
https://www.pdfen.com/pdf-a-validator didn't complain, seems valid PDF1.4 
I am not sure how advanced the checker is, maybe it could also be tried with VeraPDF  https://github.com/verapdf or https://www.datalogics.com/products/pdf-tools/pdf-checker/

zn59, 2021-03-24:
Here is my .fzz file.  Hade to zip it because the interface doesn't allow .fzz files...
My printer is a HP Photosmart 6100 series printer. The artefacts don't appear for me, just the page margin...

[Sensor board L:R.fzz.zip](https://github.com/fritzing/fritzing-app/files/6197277/Sensor.board.L.R.fzz.zip)

kittentronic, 2021-03-28:
Prints OK for me from Fritzing 0.9.6 on the Canon. Looks like this is a printer-specific issue? Should Fritzing maybe set a bigger margin, or centre the image on the page?
[image]

zn59, 2021-03-28:
Are you printing from a Mac? Otherwise I have used this printer for many many years and never noticed any problems when printing...

kittentronic, 2021-03-30:
Yes, tried it from two different Macs both running Catalina. It prints OK on a Brother laser as well.

zn59, 2021-03-30:
I'm running Big Sur on my Mac, but should not make any difference?  So are you planning any program changes to increase the margin some millimetres?  I found a workaround by placing some garbage text outside the PCB to the left...

Do you know how really big holes like 20 mm will be manufactured?  I googled around but could not found any answer. Attached an example...
[MIDI pedal main board.fzz.zip](https://github.com/fritzing/fritzing-app/files/6229260/MIDI.pedal.main.board.fzz.zip)

kittentronic, 2021-03-30:
A bigger margin sounds like a reasonable change, hopefully one of the Fritzing developers will pick that up. Re manufacturing big holes, that's a question for the fabrication house you will use isn't it?

KjellMorgenstern, 2021-04-01:
I think in all operating systems, the margin is a setting in the printer dialog, provided by the operating system, or by the printer driver. Such as the paper size like A4 vs US Letter, which may or not may be the problem. 
The observed effects on MacOS seem to be an incompatibility with transparent elements in the PDF, on also on macOS side. 
Other things discussed here are offtopic for this already broad title issue.

KjellMorgenstern, 2021-06-19:
Closing this issue in favor of #3800

closed 2021-06-19 by KjellMorgenstern