fritzing/fritzing-app #3801: error LNK2001: unresolved external symbol

Why-not-now, 2021-03-24:
## Current Behaviour
Running built Fritzing gives thiserror
> 	C:\Qt\Tools\QtCreator\bin\jom.exe -f Makefile.Debug

> 	link /NOLOGO /DYNAMICBASE /NXCOMPAT /DEBUG /SUBSYSTEM:WINDOWS "/MANIFESTDEPENDENCY:type='win32' name='Microsoft.Windows.Common-Controls' version='6.0.0.0' publicKeyToken='6595b64144ccf1df' language='*' processorArchitecture='*'" /OUT:..\debug64\Fritzing.exe @C:\Users\User\AppData\Local\Temp\Fritzing.exe.30788.3797.jom
> git2.lib(oid.obj) : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line to improve linker performance
>    Creating library ..\debug64\Fritzing.lib and object ..\debug64\Fritzing.exp
> LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; use /NODEFAULTLIB:library
> git2.lib(path.obj) : error LNK2001: unresolved external symbol __imp_GetTokenInformation
> git2.lib(path.obj) : error LNK2001: unresolved external symbol __imp_OpenProcessToken
> git2.lib(path.obj) : error LNK2001: unresolved external symbol __imp_IsValidSid
> git2.lib(path.obj) : error LNK2001: unresolved external symbol __imp_IsWellKnownSid
> git2.lib(path.obj) : error LNK2001: unresolved external symbol __imp_GetNamedSecurityInfoW
> git2.lib(path.obj) : error LNK2001: unresolved external symbol __imp_EqualSid
> git2.lib(findfile.obj) : error LNK2001: unresolved external symbol __imp_RegQueryValueExW
> git2.lib(findfile.obj) : error LNK2001: unresolved external symbol __imp_RegOpenKeyExW
> git2.lib(findfile.obj) : error LNK2001: unresolved external symbol __imp_RegCloseKey
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpConnect
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpQueryOption
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpSetStatusCallback
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_CoUninitialize
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpSetTimeouts
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_CoCreateInstance
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_CertFreeCertificateContext
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpSetCredentials
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpSendRequest
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpWriteData
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpQueryAuthSchemes
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpCloseHandle
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpSetOption
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpOpenRequest
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpReadData
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpQueryHeaders
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpAddRequestHeaders
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_CoInitializeEx
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpOpen
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_UuidCreate
> git2.lib(winhttp.obj) : error LNK2001: unresolved external symbol __imp_WinHttpReceiveResponse
> ..\debug64\Fritzing.exe : fatal error LNK1120: 30 unresolved externals

**Build:**
Version 0.9.6b? Cocoa [Qt 5.12.2]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Build Fritzing to section "RUN" in wiki
- Try running it

## Expected Behaviour
Fritzing should show up

Why-not-now, 2021-03-24:
[image]

failiz, 2021-03-24:
It is not finding the libgit2 library. Be sure to follow the instructions of the wiki, and place it correctly.

Why-not-now, 2021-03-24:
There was a bunch of similar warning while building:

> Generating Code...
> 	cl -c -nologo -Zc:wchar_t -FS -Zc:rvalueCast -Zc:inline -Zc:strictStrings -Zc:throwingNew -Zc:referenceBinding -Zc:__cplusplus -O3 -fno-omit-frame-pointer -Zi -MDd -std:c++14 -W3 -w34100 -w34189 -w44996 -w44456 -w44457 -w44458 -wd4577 -wd4467 -EHsc /Fd..\debug64\Fritzing.vc.pdb -DUNICODE -D_UNICODE -DWIN32 -D_ENABLE_EXTENDED_ALIGNED_STORAGE -DWIN64 -D_CRT_SECURE_NO_DEPRECATE -D_WINDOWS -DWIN64 -DGIT_VERSION=\"\" -DGIT_DATE=\"\" -DBUILD_DATE=\"2021-03-25T03:10:16.2337254+08:00\" -DQT_QML_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_SQL_LIB -DQT_XML_LIB -DQT_CORE_LIB -IC:\fritzing\fritzing-app-develop -I. -IC:\Qt\5.12.2\msvc2017_64\include\QtZlib -IC:\fritzing\libgit2\include -IC:\fritzing\boost_1_75_0 -IC:\Qt\5.12.2\msvc2017_64\include -IC:\Qt\5.12.2\msvc2017_64\include\QtPrintSupport -IC:\Qt\5.12.2\msvc2017_64\include\QtSvg -IC:\Qt\5.12.2\msvc2017_64\include\QtWidgets -IC:\Qt\5.12.2\msvc2017_64\include\QtGui -IC:\Qt\5.12.2\msvc2017_64\include\QtANGLE -IC:\Qt\5.12.2\msvc2017_64\include\QtConcurrent -IC:\Qt\5.12.2\msvc2017_64\include\QtNetwork -IC:\Qt\5.12.2\msvc2017_64\include\QtSerialPort -IC:\Qt\5.12.2\msvc2017_64\include\QtSql -IC:\Qt\5.12.2\msvc2017_64\include\QtXml -IC:\Qt\5.12.2\msvc2017_64\include\QtCore -IC:\fritzing\debug64 -IC:\fritzing\debug64 -IC:\VulkanSDK\1.0.51.0\include -IC:\Qt\5.12.2\msvc2017_64\mkspecs\win32-msvc -Fo..\debug64\ @C:\Users\User\AppData\Local\Temp\graphicsflowlayout.obj.12228.40953.jom
> graphicsflowlayout.cpp
> cl : Command line warning D9002 : ignoring unknown option '-O3'
> cl : Command line warning D9002 : ignoring unknown option '-fno-omit-frame-pointer'

[image]

[image]

Those folder selected were the original folders

failiz, 2021-03-24:
"Are these relevant to not being able to find libgit2?" 
No, I also have those and it works well.

The problem is the gitlib2 library. Check that you have complied it for your system and it is available. 
In may case, I have it in the path: "COMMON_FOLDER\libgit2\build64\Release" and in that folder there is a dll 
You could also try to ad that folder to the PATH environmental variable (and restart QtCreator). It should not be necessary, but it will force to find it.

Why-not-now, 2021-03-24:
[image]
I do have it

failiz, 2021-03-24:
I got a message from @Why-not-now  stating the he had the library, but you only have a lib file. You need a dll file. 
Did you use the "cmake -DBUILD_SHARED_LIBS=OFF .."?

Why-not-now, 2021-03-24:
I tried rebuilding it using the same instruction but no luck...

Why-not-now, 2021-03-24:
https://github.com/libgit2/libgit2/releases/tag/v0.28.5
My version of libgit2

Why-not-now, 2021-03-24:
`$ cd /`
`cd ~/fritzing/libgit2`
`mkdir build`
`cd build`
`cmake -DBUILD_SHARED_LIBS=OFF ..`
`cd ..`
`mkdir build64`
`cd build64`
`cmake .. -G "Visual Studio 16 2019" -A x64 -DBUILD_SHARED_LIBS=OFF -DBUILD_CLAR=OFF`
`cmake --build . --config Release`
my file is in C:/fritzing

Why-not-now, 2021-03-24:
> I got a message from @Why-not-now stating the he had the library, but you only have a lib file. You need a dll file.
> Did you use the "cmake -DBUILD_SHARED_LIBS=OFF .."?

Can you try to rebuild libgit2?

failiz, 2021-03-24:
You are making two directories and calling cmake twice. Try the command that appear si the basic installation in the gitlib page.
Also check that your architecture is x64.

Why-not-now, 2021-03-24:
> You are making two directories and calling cmake twice. Try the command that appear si the basic installation in the gitlib page.
> Also check that your architecture is x64.

So I only make build64?

Why-not-now, 2021-03-24:
Tried building only build64, error still occurs, architecture is indeed x64

Why-not-now, 2021-03-25:
> "Are these relevant to not being able to find libgit2?"
> No, I also have those and it works well.
>
> The problem is the gitlib2 library. Check that you have complied it for your system and it is [...]

Which version did you use? Maybe I should go one version down

failiz, 2021-03-25:
You have to compile libgit2 and it should generate a dll file. If there is no dll file, Fritzing will not work. Follow the instructions in the page of gitlib2 or try without -DBUILD_SHARED_LIBS=OFF
I do not think that it is a version problem, but I am using version 1.0.1

Why-not-now, 2021-03-25:
They recommended 0.28.x versions...
Lemme try both things

Why-not-now, 2021-03-25:
Maybe it is the compiler (what compiler you used? Perl-strawberry gave me error)
[image]

Why-not-now, 2021-03-25:
> They recommended 0.28.x versions...
> Lemme try both things

Btw this didn't work

failiz, 2021-03-25:
I used the Visual Studio compiler from the terminal (the own terminal of Visual Studio).

Why-not-now, 2021-03-25:
They said I got a .lib file because I did -DBUILD_SHARED_LIBS=OFF

Why-not-now, 2021-03-25:
https://stackoverflow.com/questions/22954119/linker-error-while-linking-some-windows-apis
This is the closest to the bug I can find

failiz, 2021-03-26:
Then try to not use that option. I can see that the option was added a few months ago to the wiki. Maybe it is not correct.

KjellMorgenstern, 2021-03-26:
Just a general question, why are you building Fritzing? We try to cover all platforms with our build, and we now have all versions of Windows (except XP and older) , and MacOS >= High Sierra

Why-not-now, 2021-03-27:
> Just a general question, why are you building Fritzing? We try to cover all platforms with our build, and we now have all versions of Windows (except XP and older) , and MacOS >= High Sierra

What, wdym

Why-not-now, 2021-03-27:
> 19:31:01: Starting: "C:\Qt\Tools\QtCreator\bin\jom.exe" 
> 	C:\Qt\Tools\QtCreator\bin\jom.exe -f Makefile.Debug
> Error: dependent 'C:\fritzing\libgit2\include\git2\inttypes.h' does not exist.
> jom: C:\fritzing\build-phoenix-Desktop_Qt_5_12_2_MSVC2017_64bit-Debug\Makefile [debug] Error 2
> 19:31:04: The process "C:\Qt\Tools\QtCreator\bin\jom.exe" exited with code 2.
> Error while building/deploying project phoenix (kit: Desktop Qt 5.12.2 MSVC2017 64bit)
> When executing step "Make"
> 19:31:04: Elapsed time: 00:03.
[image]

now I get this

Why-not-now, 2021-03-27:
> now I get this

This error was from my dumb mistake of using 1.1.0 libgit2
But now I have 
[image]

Why-not-now, 2021-03-27:
> 01:33:54: Starting C:\fritzing\debug64\Fritzing.exe -f "/path/to/fritzing-app/" -parts "/path/to/fritzing-parts/"...
> "Application folder translations not found"
> "translation en_US loaded 0 from ."
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "registering font family: Droid Sans DroidSans"
> "registering font family: OCRA OCRA"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "1.6 SHA empty"
> "load something"
> "checking for backups"
> "check files to load 0"
> "create empty sketch"
> "Application folder translations not found"
> "Application folder translations not found"
> "Application folder translations not found"
> QIODevice::read (QFile, "\bins\core.fzb"): device not open
> QIODevice::read (QFile, "\bins\contribParts.fzb"): device not open
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> QIODevice::read (QFile, "\bins\core.fzb"): device not open
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> QIODevice::read (QFile, "\bins\contribParts.fzb"): device not open
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "loading bin 'core'"
> "Application folder sketches not found"
> "Setting platform to Arduino"
> "updating link"
> "Setting board to Arduino BT"
> "Setting board to Arduino UNO"
> "Setting board to Arduino UNO"
> "Setting board to Arduino UNO"
> "Setting port to "
> "finish up sketch loading"
> "loading bin 'Search'"
> "Application folder fritzing-parts not found"
> "Application folder parts not found"
> "http://fritzing.org/download/feed/atom/windows-64bit/?version=0.9.6b..&sysname=Windows&kernname=NT&kernversion=10.0&arch=x86_64&sysversion=10&pid=c0a22c9aaa0403bc02e0dae9dc4a582e"
> "http check new version"
> "http check new version error Unknown error"
> "handle error"
> "handle error done"

Why-not-now, 2021-03-27:
[image]
Just a few more errors it threw at me

Why-not-now, 2021-03-27:
Fixed it by rebuilding

closed 2021-03-27 by Why-not-now