fritzing/fritzing-app #3806: Check for Updates error Windows

kgrabe, 2021-04-07:
## Current Behaviour
Dialog:
 "Fritzing is unable to check for and update new parts.
"If you want thie functionality, please enable write-permission on this folder:
C:/Program Files/Fritzing/fritzging-parts"

Changing the the write-permission doen't fix it or the write permission seems to revert to read only
[image]

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
Help | Check for Updates...

## Expected Behaviour
Dialog with update info

vanepp, 2021-04-07:
Mine looked like this:

[image]

the change needed is in the security tab here:

[image]

to

[image]

then you apparently need to reboot for the change to take effect. What would be desirable is for the installer (which I think is running as administrator) to make this change during the install if that is possible. 

Peter

kgrabe, 2021-04-07:
I changed the permissions as per  instructions (thanks!), rebooted and now it works.
Yes Peter - the installer should set the needed permissions. 
-karl

KjellMorgenstern, 2021-04-07:
If installed by the admin, then the admin should also do the updates. Giving write access to all users possibly violates security.

Another issue is: There actually are no updates past 0.9.6 yet. What are you updating to?

KjellMorgenstern, 2021-04-07:
Ah, I think we got it. If you manually trigger the update check, Fritzing will first check the write permissions, and only then check for updates. So even if there are no updates, you will get the notification about the write permission.

To fix this
- [ ] Change dialog to notify about admin access. Write access to anywhere in C:\Program Files for normal users is often not a good idea.
- [ ] Maybe delay the write permission check, first check at the server if any updates are available.

kgrabe, 2021-04-07:
Not updating, just wanted to check if there was an update which I think it did but then failed when checking for parts updates, there are no updates there either.

vanepp, 2021-04-07:
"If installed by the admin, then the admin should also do the updates. Giving write access to all users possibly violates security."

I installed (just uninstalled and reinstalled 0.9.6 to make sure) as a normal user not as admin. It didn't ask for administrator access (which is why I reinstalled, because I couldn't remember if it did when I originally installed.) As far as I know this change only gives write access to fritzing-parts and its subdirectories, to all users not all of Program files (the parent Fritzing directory still shows no modify access for users) so it shouldn't be a security problem I don't think. It would be desirable to have this setting made by the installer so automatic updates will just work without having to change the permissions though.

kgrabe, 2021-04-08:
I don't know if this is of use but the Arduino IDE seems to store Board Manager data here:
C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\hardware

with user permissions the same as System & Admin:
[image]

KjellMorgenstern, 2021-08-24:
As I understand your suggestions, this would still give all users access to all other users files. It should be not required to fiddle with permissions at all. For sure Fritzing should not open permissions by default. 

It makes a difference if you start the installer as admin or as user. Note: This is again different from the case were Fritzing asks for admin privileges during the install. The later just happens when the installer detects some updates from Microsoft that need to be installed (msvc dll and similar).

To go back to the Arduino15\packages\arduino\hardware example: If the admin creates a file in this directory, I think a user will still not be able to overwrite it.

KjellMorgenstern, 2025-03-06:
We have few core parts updates recently. Since the parts update mechanism caused quite some trouble, it is much more efficient to just distribute the parts together with app updates. 
Closing in favor of #4024 .

closed 2025-03-06 by KjellMorgenstern