fritzing/fritzing-app #3807: regression: Images on copper layers missing from gerbers

deshipu, 2021-04-07:
## Current Behaviour
In Fritzing 0.9.6 and 0.9.5, images placed on the copper layer do not appear in generated Gerbers:
[image]

In earlier versions (0.9.3 is what I tested), the images appear correctly:
[image]

<please attach screenshot if possible>
**Build:**
Version 0.9.6

**Operating System:**
Linux

**Steps to reproduce:**
Create a PCB with an image on a copper layer.
Export the PCB as gerbers.
See the image missing from the exported gerbers.

## Expected Behaviour
The image should be included in the exported gerbers.

KjellMorgenstern, 2021-04-08:
Confirmed. 
I tested by exporting a production PDF of the Ocarina v1 on hackaday, which users similar images.
0.9.4 has the pattern on copper top
0.9.6 does not have the pattern

deshipu, 2021-04-08:
If that helps, I also noticed that if you re-do the ground fills, it ignores the images in 0.9.5 and 6, while it goes around them in 0.9.3.

KjellMorgenstern, 2021-04-14:
This is resolved by adding 'e' and 'E' to svg token list. It is not a svg path command, but it is used (and valid svg) in scientific notation of numbers. Pathc upcoming.

commit a7a95c1b (2021-05-24): Fixed #3807 without breaking #3647 by fixing svg lexer float exponent separation bug.
commit 0e398a62 (2021-10-06): Merge develop

Silence empty polyline assertion

Replace arg(QString).arg(QString) with arg(QString, QString)
Fix more warnings

Upload to fab, happy flow passes

Fix build warning on windows

Refactor renderthing into own file

Fix quirk with disabled rotate and align submenus

addToBin now also re-enabled

Always center mouse when draging from the parts bin.

Fixes some quirks with rapid mouse movement or when leaving and re-entering the window.

Fixed #3807 without breaking #3647 by fixing svg lexer float exponent separation bug.

Added simplified unit test case for pathlexer clean of float [... cut]
closed 2021-05-24 by KjellMorgenstern