fritzing/fritzing-app #3808: Add

kgrabe, 2021-04-09:
## Problem
'Backpace' should delete objects as well as the numberic keypad 'Del' key.
It's a large key, easy to find, and programs like LibreOffice Draw use it this way.
The 'Del' key only works when the 'Num Lock' is off, novice users might not know that.

[image]
Related bug: "Unable to Delete Notes from Keyboard #3714"

## Proposed Solution
Enable 'Backspace' to delete objects: parts, text, notes etc.

failiz, 2021-04-10:
"Unable to Delete Notes from Keyboard #3714" is not related to this. #3714 does not work with any delete key.
But I agree that the backspace could be a good addition.

KjellMorgenstern, 2021-04-10:
I think this is invalid.
We already have the 'Delete' key (right of Backspapce) . It has the same meaning as 'Del' on the Numpad, in all applications I think, not only in Fritzing.

I think "Backspace" should not delete the complete part. But it could 
* remove bendpoints from a wire
* disconnect a wire without deleting it (not so sure if that will be usefull or just cause complications)
* Especially when dealing with Notes, it already deletes the character before the cursor. It should *not* delete the part/complete note

vanepp, 2021-04-10:
"disconnect a wire without deleting it (not so sure if that will be usefull or just cause complications)"

I think this could be very useful if it also moved the end of the wires 0.1in (which may be hard!). The use case is delete minus which deletes the part but leaves the traces. The problem is with a pad with wires on both sides like this, I find it almost impossible to move just the end of one of the wires. Here I want to replace the two headers with the Arduino Micro part. So I delete minus the first header (selected in this image)

[image]

after the delete minus it looks like this. Here I clicked on the right most wire and dragged it to the right. Instead of moving the wire to the right, Fritzing appears (from the actions in the undo window) to have joined the two wires which isn't what I want because I need to drag the end of the wire to connect it to the new part once I add it. 

[image]

Here I used delete wire to breakbpoint then moved the right most wire and recreated the bendpoint on the left most wire to create the condition that I want and that I would like the backspace key too do!

[image]

The wires circled in green can now be dragged on to the new part pin to reconnect them. The problem I see may be determining which way the wire needs to move the 0.1in, it will depend on the orientation of the wire I expect and may be to complex to implement.

failiz, 2021-04-10:
@KjellMorgenstern , the delete and backspace work as expected when the text has been selected. The delete key removes the next character after the curse, the backspace key deletes the previous character. When the note has been selected (not the text of the note), the delete key removes the note but the backspace key does not.  It could be difficult to observe this as there is a bug that avoids to select the note object and the text is always selected ( #3714 and #3281). You can observe this if you launch fritzing and the launch a new instance from file->new (the the behavior is correct).

I am not opposed to implement different behaviors for the backspace and delete key in wires or other objects. However, I do not see any other possible behavior than deleting the note in this case. I think that there are programs that use delete or backspace for the same action (deleting objects).