fritzing/fritzing-app #3809: Silkscreen path not appearing in gerber output

vanepp, 2021-04-10:
## Current Behaviour

Looks like there are still issues with paths in 0.9.6 (although it is a lot better!) In this part 

[seeed-grove-i2c-hub.fzpz.zip](https://github.com/fritzing/fritzing-app/files/6290387/seeed-grove-i2c-hub.fzpz.zip)
(just remove the trailing .zip to get the .fzpz file) the path renders correctly in Fritzing 

[image]

but not on the silkscreen layer in the gerber output 

[image]

**Build:**
Version 0.9.6 
(bCD-175-0-8a1e0682 2021-02-21) 64 [Qt 5.12.10]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Load the seeed-grove-i2c-hub.fzpz in to Fritzing
- File->Export->For Production->Extended Gerber
- Display the resulting geber silktop.gbl file in a gerber viewer (gerbv in this case) 

## Expected Behaviour
The path should appear on the silkscreen in the gerber export.

KjellMorgenstern, 2021-04-17:
Same cause as #3807

vanepp, 2021-04-17:
I just last night found a different example, in the copper layers this time. Hopefully it is the same cause not a new bug! First the two parts (just remove the trailing .zip to get the .fzpz)

[SK6812 Mini-E RGB LED-broken.fzpz.zip](https://github.com/fritzing/fritzing-app/files/6329662/SK6812.Mini-E.RGB.LED-broken.fzpz.zip)

[SK6812 Mini-E RGB LED-fixed.fzpz.zip](https://github.com/fritzing/fritzing-app/files/6329663/SK6812.Mini-E.RGB.LED-fixed.fzpz.zip)

geber output from 'SK6812 Mini-E RGB LED-broken.fzpz'

[image]

Fritzing pcb view for 'SK6812 Mini-E RGB LED-broken.fzpz'

[image]

geber output from 'SK6812 Mini-E RGB LED-fixed.fzpz'

[image]

Fritzing pcb view for 'SK6812 Mini-E RGB LED-fixed.fzpz'

[image]

The difference between 'SK6812 Mini-E RGB LED-broken.fzpz' and 'SK6812 Mini-E RGB LED-fixed.fzpz' is that I noted the x and y coords of the top 2 pads, then deleted them. Then I duplicated the bottom 2 pads (that rendered) and moved them to the same x y coordintes as the top two pads. Then the pads rendered correctly. So the d values of the top 2 pads may provide a clue as to where the bug is. Or better yet it may be cured along with the original case here!

Peter

closed 2021-07-25 by KjellMorgenstern