fritzing/fritzing-app #3810: Support more Linux Distributions/Versions

Bengt, 2021-04-17:
## Problem

Currently, Fritzing only supports one distribution of Linux, namely Ubuntu:

<https://fritzing.org/releases/0-9-6>

While there was support for two Ubuntu releases in the past, this has now ceased:

<https://fritzing.org/releases/0-9-4>

Compare that to the Ubuntu release schedule, which keeps two to three LTS versions under (extended) support at all times:

<https://en.wikipedia.org/wiki/Ubuntu_version_history#Table_of_versions>

Further, users in this issue tracker are using Fritzing under Linux distributions other than Ubuntu:

-    [120 issues regarding Ubuntu](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+Ubuntu)
-    [47 issues regarding Fedora](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+Fedora)
-    [32 issues regarding Debian](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+Debian)
-    [9 issues regarding openSUSE](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+openSUSE)
-    [5 issues regarding CentOS](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+CentOS)

Note that apart from Ubuntu users, these issues were filed by people who compiled Fritzing themselves. So the latter numbers can be considered a lower bound of demand from these user groups. Given a precompiled build and official support, one could expect a considerably larger addressable user base.

Also note that the Linux user base seems the second largest, while currently having less support than the other operating systems:

-   [363 issues regarding Windows](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+Windows+OR+Windows-10+OR+Windows10)
-   [317 issues regarding Linux](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+Linux+OR+Ubuntu+OR+fedora+OR+debian+OR+openSuse+OR+linux-mint)
-   [219 issues regarding macOS](https://github.com/fritzing/fritzing-app/issues?q=is%3Aissue+macOS+OR+mac-OS+OR+OS-X+OR+OSX+OR+Catalina+OR+Big-Sur)

## Proposed Solution

Fritzing should follow the release model of current Linux distributions as closely as possible. This means ideally supporting as many distributions as possible and their versions currently supported by their respective distributors. To prove that this is possible in principle, I created such builds for a couple of distributions:

<https://gitlab.com/Ka-Ching/Ka-Ching.Auto-Builds>

End users can use these builds now to run Fritzing on their machines. However, potential end users need to find them, which is unnecessarily difficult. To make Fritzing immediately available to a broader user group, these builds should be linked hosted at the download page of Fritzing.

Testers have the same problem, in that they might not know about these automatic builds. They might want to install the latest master version of this repository, without compiling the code themselves. To that end, they need guidance on how to download these versions. Such guidance could be a wiki article on testing the application or a paragraph in the README file.

Developers need to make sure their changes do not break compatibility with any version of any distribution. Continuous integration can help them by testing their changes as closely as possible. Currently, the automated builds are driven by a weekly build of the base images of their build environments. Maybe there should be a second trigger coming from GitHub to integrate the latest changes with all operating system versions. This should be possible using a Web Hook:

<https://docs.github.com/en/developers/webhooks-and-events/about-webhooks>

KjellMorgenstern, 2021-04-17:
We have been working on an AppImage, only problem (not a real one) remaining is icons and file associations. Release will follow soon.

Bengt, 2021-04-17:
> We have been working on an AppImage, only problem (not a real one) remaining is icons and file associations. Release will follow soon.

Hm, doesn't app-image prohibit installation of the applications into the operating systems? I am not sure about what the most elegant way of deployment is. That is why I stuck to the current method of delivering an archive of files for all use cases.

> PS: The github statistic is strongly biased.

Do you have better statistics? Maybe about the package downloads?

stefan123t, 2021-04-19:
[image]
[image]
I found these on LinuxMint 21 in /usr/share/pixmaps and /usr/share/app-install/icons/

Bengt, 2021-04-20:
Note that there are lots of  distributions providing Fritzing packages:

<https://pkgs.org/download/fritzing>

So, there is value in ensuring compatibility using a continuous integration, as I set up. - Regardless of whether there is a distribution-agnostic image available.

closed 2021-07-25 by KjellMorgenstern