fritzing/fritzing-app #3812: Trying to build Fritzing but issue: "Fritzing does not build with Qt 5.8 or earlier. 5.12 is recommended."

t-scale, 2021-04-21:
## Current Behaviour

I'm trying to build Fritizing after this guide:
https://siytek.com/build-fritzing/
Qt version is higher than 5.12 but i get the above error.

[image]
[image]

**Build:**
Qt Creator 4.14.2
Based on Qt 5.15.2 (Clang 11.0 (Apple), 64 bit)

Built on Mar 19 2021 09:03:22

**Operating System:**
MacOS 10.14.6 (Mojave)

**Steps to reproduce:**
- Install Qt Creator with installer
- load phoenix.pro from fritzing-app into Qt Creator

## Expected Behaviour
It should build Fritzing.

closed 2021-04-22 by KjellMorgenstern

t-scale, 2021-04-22:
why is this invalid?

t-scale, 2021-04-22:
my bad i'd just missed one step...

leao-nardo, 2021-06-28:
> my bad i'd just missed one step...

im having the same error how did you fixed ?

leao-nardo, 2021-06-28:
i got this error when using qt 6.1.1 and 6.1.2 when i installed the version Qt 5.15.2 it didnt gave the error

WeepingRaven, 2021-07-26:
> my bad i'd just missed one step...

what step?

StevenLovell, 2021-10-04:
There's a special place in hell for people who fix there own problem then don't post the fix.

KjellMorgenstern, 2021-10-04:
@StevenLovell There isn't anything to fix. I think the problem might be sitting in front of your screen.

StevenLovell, 2021-10-04:
> @StevenLovell There isn't anything to fix. I think the problem might be sitting in front of your screen.

Probably.

> i got this error when using qt 6.1.1 and 6.1.2 when i installed the version Qt 5.15.2 it didnt gave the error

This sorted it for me.

fanatic007, 2022-06-08:
@leao-nardo @StevenLovell How do I use the extract to run the installer? Or better, how can I get a .dmg ..?