fritzing/fritzing-app #3813: fzz file won't open on Windows but does on Linux

vanepp, 2021-04-26:
## Current Behaviour

On Windows this file name (delete the trailing .zip to leave the .fzz file) won't load, but on Linux
it does. 

[HIZLI Çİ.fzz.zip](https://github.com/fritzing/fritzing-app/files/6378483/HIZLI.CI.fzz.zip)

[image]

(note another bug, the error message is lacking errno so we get a %d instead of the error code!)

reducing the file name by 1 letter makes it load on Windows (file contents are identical, only the file name is different by 1 character!)

[HIZLI Ç.fzz.zip](https://github.com/fritzing/fritzing-app/files/6378524/HIZLI.C.fzz.zip)

[image]

The difference appears to be one unicode character, the odd part is it works on Linux just not Windows (don't know about the Mac)

$ ls 'HIZLI Ç.fzz' | od -x
0000000 4948 4c5a 2049 87c3 662e 7a7a 000a
0000015

$ ls 'HIZLI Çİ.fzz' | od -x
0000000 4948 4c5a 2049 87c3 b0c4 662e 7a7a 000a
0000017

**Build:**
Version 0.9.6 
(bCD-175-0-8a1e0682 2021-02-21) 64 [Qt 5.12.10]

**Operating System:**

Windows 10

**Steps to reproduce:**
- Attempt to load HIZLI.CI.fzz in to Fritzing. Fails with a zip error message
- Attempt to load HIZLI Ç.fzz in to Fritzing, loads as expected.

## Expected Behaviour

Attempt to load HIZLI Ç.fzz in to Fritzing, loads as expected.

KjellMorgenstern, 2024-09-04:
This likely was fixed by Qt updates or similar, can't reproduce it anymore.

vanepp, 2024-09-12:
@KjellMorgenstern I agree, I just tried on 1.0.3 and the .fzz loads fine so something looks to have fixed this so I'll close this issue.

closed 2024-09-12 by vanepp