fritzing/fritzing-app #3814: Project ERROR: libgit2 library not found /libgit2/build32/Release

fscii, 2021-04-29:
## Current Behaviour
When I get to the step in QT maker of changing the paths in the run section then press the green play button, I get the following
problem - for some reason its looking for libgit2/build32/Release instead of libgit2/build64/Release which I have.

Compile Output:
13:24:56: Starting: "C:\Qt\5.15.2\mingw81_64\bin\qmake.exe" C:\Users\fscii\projects\fritzing\fritzing-app\phoenix.pro -spec win32-g++ "CONFIG+=debug" "CONFIG+=qml_debug"
Project MESSAGE: target arch: 
Project MESSAGE: found libgit2 include path at C:/Users/fscii/projects/fritzing/fritzing-app/../libgit2/include
Project ERROR: libgit2 library not found in C:/Users/fscii/projects/fritzing/fritzing-app/../libgit2/build32/Release
13:24:56: The process "C:\Qt\5.15.2\mingw81_64\bin\qmake.exe" exited with code 3.
Error while building/deploying project phoenix (kit: Desktop Qt 5.15.2 MinGW 64-bit)
When executing step "qmake"
13:24:56: Elapsed time: 00:00.

General Messages:
Project MESSAGE: target arch: x86_64
Project MESSAGE: found libgit2 include path at C:/Users/fscii/projects/fritzing/fritzing-app/../libgit2/include
Project MESSAGE: found libgit2 library in C:/Users/fscii/projects/fritzing/fritzing-app/../libgit2/build64/Release
Project MESSAGE: 
Project MESSAGE: Using fritzing boost detect script.
Project MESSAGE: using ../boost_1_76_0 boost environment
Project MESSAGE: target arch: x86_64
Project MESSAGE: found libgit2 include path at C:/Users/fscii/projects/fritzing/fritzing-app/../libgit2/include
Project MESSAGE: found libgit2 library in C:/Users/fscii/projects/fritzing/fritzing-app/../libgit2/build64/Release

As you can see for some reason, General Messages finds libgit2/build64/Release but the Compile goes sideways looking for
libgit2/build32/Release.

How do I fix this?

**Build:**
Using Windows10 64 bit.
QT 5.15.2 within QT creator 4.14.2 (mingw version)
Visual Studio 2019 (haven't reached this step yet)

**Operating System:**
Windows10

**Steps to reproduce:**
Following this: (and used Visual Studio to make libgit2 v0.28.5)

## Expected Behaviour
Compiling would be nice :)

vanepp, 2021-05-03:
Assuming I am recalling correctly, I don't think mingw will compile Fritzing on Windows. I seem to recall them specifying that you need to us Visual Studio to build libgit2. There are build instructions on the forum here which may help:

https://forum.fritzing.org/t/build-instructions-for-linux-and-windows/5528

If you understand how to configure new compilers in QT (I don't :-) ), you can add Visual Studio, I build Visual Studio first then install QT as then QT configures the VS compiler the way it wants it. The settings I used on VS 2017 (probably) are in the instructions above, I expect it will be much the same on VS 2019 but I haven't upgraded to that yet.

fscii, 2021-05-03:
> Assuming I am recalling correctly, I don't think mingw will compile Fritzing on Windows. I seem to recall them specifying that you need to us Visual Studio to build libgit2. There are build [...]

I did build libgit2 via Visual Studio 2019 command line however there was an error on this page:
https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing

It says use option -DBUILD_SHARED_LIBS=OFF (which generates git2.lib not git2.dll) so there's that.  I changed it to ON.

That being said, I had installed QT before installing Visual Studio so it had defaulted to mingw.  In the uninstaller there's options to reconfigure so I had it pull out mingw and go to Visual Studio as you suggested and it worked.

Much thanks!

closed 2021-05-03 by fscii