fritzing/fritzing-app #3816: no decimal values for drill holes possible

electronic-cosmos, 2021-05-02:
## Current Behaviour
Text
<please attach screenshot if possible>
**Build:**
Version 0.9.6 
(bCD-175-0-8a1e0682 2021-02-21) Cocoa [Qt 5.15.2]

**Operating System:**
<BIG sur 11.2.3>

**Steps to reproduce:**
Put a Drill hole to PCB and change the diameter of the hole.
Wondering that for only natural numbers and no commas are possible for hole and copper.

## Expected Behaviour
comma values are neccessary for drillhole generation.

failiz, 2021-05-02:
Probably, duplicate of #3080
What's Fritzing's language? Or OS's language?

failiz, 2021-06-08:
The reported did not add extra information in one month. I assume that this a duplicate of #3080 and I would close the issue.

failiz, 2021-11-02:
Any reason for not closing this issue as it looks like a duplicate?

closed 2021-11-02 by KjellMorgenstern