fritzing/fritzing-app #3818: Fritzing compiles/runs in QT but fails Visual Studio 2019 release

fscii, 2021-05-03:
Since it runs in QT, libgit2.dll and everything should be fine.  I've updated the path in the release_fritzing.bat file and as you can see from the screenshot libgit2 is found but it bombs at the linker stage.  Help please.

<please attach screenshot if possible>
[image]

<e.g. Version 0.9.6 Cocoa [Qt 5.15.2] Visual Studio 2019

**Operating System:**
<Windows 10>

## Expected Behaviour
Text

KjellMorgenstern, 2021-05-03:
What are you trying to do? Is there any particular issue in Fritzing you want to work on?

fscii, 2021-05-03:
I am trying to compile a release version in Windows.  It bombs on the final step.

Side note:
I did come across one error in:
https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing

Under windows:
cmake .. -G "Visual Studio 16 2019" -A x64 -DBUILD_SHARED_LIBS=OFF -DBUILD_CLAR=OFF

Shared Libs must = ON or the resulting file is git2.lib rather than git2.dll

I solved that to get it to work in QT, now I just need to see how to fix this problem in building the final release.

KjellMorgenstern, 2021-05-03:
Well, building a release is quite different. Thanks for the change in the wiki, I guess it is easier to use the dll, especially with Qt Creator. However, we can not offer help just for the purpose of building Fritzing.

closed 2021-05-03 by KjellMorgenstern