fritzing/fritzing-app #3824: Different icon for macOS (Big Sur)

moritzmhmk, 2021-05-07:
The current flat Fritzing icon looks really nice on its own but feels a bit out of place on macOS (size- and stylewise).

I designed this vector icon for Fritzing to fit the new icon style of macOS Big Sur in the hope that you may find it useful.

[image]

KjellMorgenstern, 2021-05-17:
One problem first, Fritzing itself as well as Safari can not render this correctly, both do not support fe.. elements. Didn't test in safari, just read the specs. Fritzing renders it black and white.

Minor note: I think these elements are zero bboxes and can be removed:
```
   <!-- vertical lines -->
      <line x1="-76" y1="-206" x2="-76" y2="206" fill="none" />
      <line x1="0" y1="-206" x2="0" y2="206" fill="none" />
      <line x1="76" y1="-206" x2="76" y2="206" fill="none" />
      <!-- horizontal lines -->
      <line x1="-206" y1="-76" x2="206" y2="-76" fill="none" />
      <line x1="-206" y1="0" x2="206" y2="0" fill="none" />
      <line x1="-206" y1="76" x2="206" y2="76" fill="none" />
```

Further, the graphic would have to adapt for fz, fzp, fzz, fzpz and fzb file types.

The shadow line at the bottom is part of the fritzing style, i'd prefer to keep it. Not sure how that interacts with BigSur design guidelines.

moritzmhmk, 2021-05-17:
The guideline for the [App Icon](https://developer.apple.com/design/human-interface-guidelines/macos/icons-and-images/app-icon/) does not seem to conflict with the bottom line. In fact, Apple uses lines at the top of the calendar and notes icons. 

Originally I replaced the bottom line by a gradient as can be found on many Big Sur icons:
[image]

This is an adaptation of the dimensions of the current icon:
[image]

This is a version with a larger bottom line to move the "cut" out of the rounded corners.
[image]

The guideline on [Document Icons](https://developer.apple.com/design/human-interface-guidelines/macos/icons-and-images/document-icons/) states that its possible to "supply any combination of background fill, center image, and text" and has an example that looks rather flat. I believe that using the current (flat) style works well for document icons on macOS Big Sur.

Support of certain SVG features doesn't matter since for macOS the icons need to be converted to PNG anyway, but I did post PNG versions above to make it easy for anyone to view in the web browser. The icon will of course eventually be contributed as a vector graphic.