fritzing/fritzing-app #3826: pcb view bottom layer selected puts parts on bottom.

vanepp, 2021-05-09:
## Current Behaviour

I'm not sure if this is a bug or expected (if in my view undesirable) behaviour.  

[image]

if I then select bottom layer like this (expecting to  only route traces on the bottom layer)

[image]

then drag in a new part, the part gets oriented as inserted from the bottom (rather than on top as is normal.) I expect that selecting bottom layer routes traces on the bottom layer. I don't expect it to place a new part on the bottom layer (which will reverse the pins and if as most people will, you insert the part from the top, it will be incorrectly connected!)

[image]

**Build:**

Version 0.9.6 (bCD-175-0-8a1e0682 2021-02-21) 64 [Qt 5.12.10]

**Operating System:**
Windows 10

**Steps to reproduce:**
- See current behaviour 

## Expected Behaviour

Part should be placed on top as is usual unless the pcb layer  field in Inspector is set to bottom.

KjellMorgenstern, 2021-05-09:
It is fine to have parts on both sides of the PCB. If you insert a SMD, not a THT part, it should also only create pads on the bottom layer. 

The feature to have parts on the bottom layer is however missing in the breadboard view, you can't fully use double sided stripboards

vanepp, 2021-05-12:
I agree it is fine to have parts on both sides of the board. The issue here is that selecting to only route traces on the bottom of the board in the user interface has as an unexpected (and in my view unwanted!) side effect of also placing parts on the bottom of the board in Inspector. This is not what I expected to happen and likely not what others expect to happen. Placing a part on the board when the route traces on the bottom setting is set is not very noticeable unless you look at Inspector or look at the part label which is on the bottom silk screen layer and thus reversed, rather than the top. This becomes a problem for any part that is polarized (such as ICs and most modules) if inserted from the top, because the pins will be reversed possibly reversing power and destroying the chip. It will be routed correctly because Fritzing knows where the pins are, but It definitely won't work as expected unless you change the chip or module to mounting on the bottom of the board which you likely didn't intend. What I am suggesting is that the orientation of the part placement in Inspector not change when you set the route traces only on the bottom layer in the user interface only if you change the setting in Inspector to do so (not if you select to route traces on the bottom layer!)

KjellMorgenstern, 2021-05-17:
I can not follow. 
Selecting the bottom layer like you described is exactly the way to put parts to the bottom layer. Why do you think this is a bug?
When the bottom layer is selected, it would be unexpected if parts are added at the top layer.

The orientation of the view is changed with the "view from above" or "view from below" setting.

vanepp, 2021-05-18:
Then I'm not explaining it correctly. Here is another try. This image is correct, selecting route traces on the bottom layer only causes routing on the bottom as expected. 

[image]

Now however dragging a new resistor from core parts in to the sketch puts the resistor on the bottom layer in Inspector. This is unexpected and I think incorrect. Only changing the top to bottom in Inspector should cause the part to be placed on the bottom of the board, not selecting the traces only route on the bottom layer but in the UI. The two (at least in my view) shouldn't be connected. Changing where traces are routed shouldn't be affecting which side of the board new parts are placed in Inspector but that is what is happening now. The only change from when I dragged the first two resistors in to the sketch is that route bottom layer only icon was selected. I didn't change the default top in Inspector to bottom as it is now, that was done by the select bottom layer only button circled in yellow and I don't think it should have been. 

[image]

KjellMorgenstern, 2021-05-22:
Again I think this is expected and correct behavior.
You select the "bottom layer" to be the active layer. Now, traces and parts will be inserted at the bottom layer. To stay with the example, the R3 is inserted at the bottom layer, because the bottom layer is active when R3 is pulled onto the board.
This shows in the inspector, as the pcb layer placement is set to "bottom" there. 

Maybe this becomes clearer with a slightly different use case: Insert several SMD parts on the bottom layer: You select the bottom layer to be active, and drag and drop the SMD parts onto the board. The all appear at the bottom layer.

Still hard to follow your thoughts, it seems crystal clear to me that parts should appear at the active layer, not anywhere else.

vanepp, 2021-05-23:
I guess it is a case of expectations. I expect changing the window to bottom layer to change only trace routing to be only on the bottom layer, not where the components get placed. I can see your argument that the bottom layer has been selected so components should be placed there, but to me that is counter intuitive, and I expect causes problems for new users. The normal case is that components get placed on the top layer unless you specifically change the layer field in Inspector. This originally came up in the forums where someone had a resistor (which didn't matter because it is non polarized) on the bottom layer which he did not intend and did not know how he had gotten it there. If that had been an IC or another part where orientation matters, it would have caused a problem if not noticed (and I doubt it would have been noticed) when the IC was installed in the top of the board. The pins will be in the wrong order and the board won't work. Most ot the time components are placed on the top of the board, it is fairly rare (at least in my experience) to want to place components on the bottom of the board, but the current behaviour makes that easy to do by accident.I guess we somehow need to get other people's opinion of what the expected action is, perhaps I am the only one who thinks this is the incorrect action. Perhaps I'll bring it up in the forums and see what other folks think.

KjellMorgenstern, 2021-05-23:
So, for sure it is not a regression. 
Changing the behavior will irritate whoever is using it, especially when setting SMD parts. 
I think we had/have a different ticket, were the inserted part is a board. Maybe there are issues with specific parts?
We only know of the (very few?) cases where people got it wrong. It should be looked into those cases to see what caused the misunderstanding. 
It would be interesting to collect ideas how to make new users aware:

- Give a one time notice when top view is used with bottom active layer

- Give a one time notice when inserting a THT part on the bottom layer

- Mirror the part while it still held with the mouse pointer. This should be done anyhow, and also might help to clearify which one is the active layer

Also related:
- Should THT parts get a copper donut on both sides, or just the active side?

vanepp, 2021-05-25:
"Give a one time notice when inserting a THT part on the bottom layer" 

This may be the best solution. It doesn't look like anyone else thinks this is a big issue. This would at least give an indication that the part is being installed on the bottom layer which may not be what was expected. 

"Should THT parts get a copper donut on both sides, or just the active side?"

As far as I know there should always be pads on both sides  as I believe there needs to be copper on both sides to get the plating through in pcb. Verifying that with the Fab folks would be a good bet though as I'm not a pcb expert.

KjellMorgenstern, 2024-02-06:
Closing in favor of #4113

closed 2024-02-06 by KjellMorgenstern