fritzing/fritzing-app #3829: macOS, building against qt 5.15.2, Preferences Menu is not enabled

kencu, 2021-05-15:
## Current Behaviour
When I build Fritzing on MacOS (10.13 and 10.14 tried) against qt 5.15.2, the Preferences Menu is not enabled. 

[image]

[image]

<please attach screenshot if possible>

Version 0.9.6 (8a1e0682d7846f6384ae45373961ed693b4c55d2 20210221) Cocoa [Qt 5.15.2]

**Operating System:**
macOS 10.13 and macOS 10.14

Downloading the prebuilt version <fritzing-3d61c58421bdb63ca903bb5d11310a257f1ec0ed-develop-548.mojave.10.2.dmg> shows a normally-working Preferences menu.

[image]

**Steps to reproduce:**
- checkout fritzing-app 0.9.6. build against qt 5.15.2 on macOS 10.13 or 10.14.

## Expected Behaviour
The Preferences menu will be enabled, and function.

kencu, 2021-05-15:
I tried building the current tip of the development branch a3d8cab74c48a67795d71c199c378317fe3aa633 and I see the same greyed-out Preferences menu doing that as well.

kencu, 2021-05-16:
building against qt 5.11 gives a working Preferences menu.

KjellMorgenstern, 2021-05-21:
Preferences menu works fine with 5.15.2 , see our official 0.9.6 build.

closed 2021-05-21 by KjellMorgenstern

kencu, 2021-05-21:
cool. I must just have a bad download of qt 5.15.2 on those systems.

I'll reinstall it I guess.

My error.

kencu, 2021-05-22:
As a MacPorts maintainer who has built and installed a great many qt4 and qt5 applications, I would like to deploy this application on MacPorts, where there are many thousands of users, however this issue prevents that from happening.

MacPorts has a "policy" of requiring source builds rather than installing prebuilt official builds, so I can just copy in your build from the dmg.

This same error happens on multiple machines, so either all the qt installs on MacPorts are broken somehow (possible -- I did the last set of qt5 updates on MacPorts, and I don't have certificate of perfection on my driver's license), or there is some issue that only shows up here. No other qt5 software appears to show this issue that I have seen so far.

If I do sort it out, I'll report back here what the eventual fix was.