fritzing/fritzing-app #3832: Moving connected parts

edson-acordi, 2021-05-22:
## Problem
Describes from a user perspective the current state and its shortcomings.

Hi, I would like to know if there is the possibility to configure the Fritzing to move together with the component (part), all the existing connections. It is frustrating when you have a project and need to do some modifications, but the connections between the parts and the wires are lost when you move the parts on the breadboard.

## Proposed Solution
Describes from a user perspective the ideal state of the implementation

It would be very helpful to have an option to change the way a component (part) moves. For example, having an option to move only the part and other option to move the part and all the connected wires in it in a way that you do not lose the connections.

edson-acordi, 2021-05-29:
Selecting the IC to move:
[image]

After the part was moved (the wires connected to the IC was lost):
[image]

KjellMorgenstern, 2021-05-29:
Not a bug. That is exactly how the IC is intended to move.

edson-acordi, 2021-05-30:
Ok, but this behaviour of Fritzing is terrible when you need to do some modifications to an existing project. When the part is moved, you lose the connections!

KjellMorgenstern, 2021-06-03:
The part is connected to the breadboard. I don't think you want do move the breadboard along with it.
Rerouting the wires which you have manually placed before would be a different feature, and would be at least equally surprising or annoying to other users.

As mentioned before, the logical connection (aka ratsnest line) is kept if the part was connected or even routed in the pcb or schematic layer.

closed 2021-06-03 by KjellMorgenstern