fritzing/fritzing-app #3833: Platform, Board and Port menus are a barely readable white on grey on Macos

BruceH2020, 2021-05-29:
Macos Big Sur 11.3; Fritzing 0.9.6 downloaded from the Fritzing website.

The selection menus for choosing platform, board and port are a very hard to read white text on light grey background. Please see attached screenshot.

[image]

(I don't see a way to select a severity when submitting this to Github but "cosmetic" is fine - would be nice to get something more legible, but the world won't end.) :-)

KjellMorgenstern, 2021-05-29:
Which color scheme do you have configured on MacOS?

BruceH2020, 2021-05-29:
I'm using light. But changing to dark and restarting Fritzing made no difference.

KjellMorgenstern, 2021-07-13:
Not really fixed. It seems we have now a problem with dark letters on dark background in some Windows configurations.

commit 0e398a62 (2021-10-06): Merge develop

Silence empty polyline assertion

Replace arg(QString).arg(QString) with arg(QString, QString)
Fix more warnings

Upload to fab, happy flow passes

Fix build warning on windows

Refactor renderthing into own file

Fix quirk with disabled rotate and align submenus

addToBin now also re-enabled

Always center mouse when draging from the parts bin.

Fixes some quirks with rapid mouse movement or when leaving and re-entering the window.

Fixed #3807 without breaking #3647 by fixing svg lexer float exponent separation bug.

Added simplified unit test case for pathlexer clean of float [... cut]
closed 2021-10-06 by KjellMorgenstern