fritzing/fritzing-app #3836: Parse error (2) at line 1, column1:

itswally1, 2021-06-03:
## Current Behaviour
Text
<please attach screenshot if possible>
**Build:**
fritzing-0.9.6-175.64
**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1 Use installer from website for fresh install
- Step 2 the red splash screen says Fritzing Beta
- the text box appears on the screen saying loading
- bars appear and indicate 14% then indicates program is not responding
- hangs at that point.
Went back to version 0.8.7b pc and the load and run worked.
Tried other back Beta versions  and other versions with same problem until I got to this fritzing-0.8.7b pc version.
Granted I did not try all versions but made jumps until I got to this version and it worked.
Also I tried the fix application and it still had the same error.

Is there a corrupt part file in the default load?
Please fix the corrupted versions for Windows 10

## Expected Behavior
The software would complete the load and run after fresh load

KjellMorgenstern, 2021-06-04:
Can not reproduce.

vanepp, 2021-06-07:
The likely answer is 

"bars appear and indicate 14% then indicates program is not responding
hangs at that point."

Fritzing was accessing github to uppdate the parts database (if needed.) Interrupting it at this point corrupts the user directories (which are not cleared by a reinstall.) The following should clear your problem, and when you start Fritzing even if it says "not responding" wait a couple of minutes to see if it recovers (it should once it finishes checking github.) 

There are two user directories (with your parts and the parts database) which don't get touched during an install (to not affect your sketches during upgrades). On Windows they are in

c:\users\username\AppData\Fritzing\roaming\Fritzing (which is a hidden directory so you need to enable hidden directories in explorer) and

c:\Users\username\My Documents\Fritzing (where username is your windows id)

If you don't have any parts or sketches you want to keep you can just delete those two directories and Fritzing will recreate them, or you can move them aside by renaming them if you want to keep something in them.

linux

~/Documents/Fritzing/parts
~/.config/Fritzing 

Mac

/Users/username/Documents/Fritzing/parts
~/.config/Fritzing

KjellMorgenstern, 2021-06-08:
Fritzing should be improved so it
1. does not break the part db if github times out
2. can recover a broken part db if github is available

itswally1, 2021-06-08:
Thanks for the comments and advice. The theme of the answers seem to say a timeout for gethub is the problem. Why does it work for an earlier version ( 0.8.7b pc ) not the latest version?( I am on fiber and my network has a 100K bandwidth) Does the newest version have a larger parts file than the older version causing the timeout issue? Seems strange the older version works just fine. That is why I think it is a versioning issue.
Happily making circuits for fun and entertainment...

KjellMorgenstern, 2021-06-08:
Old versions of Fritzing don't use github, they come with a fixed set of parts that is difficult to update.

itswally1, 2021-06-08:
Thanks for the information on the downloading process for older versions.
I guess github has traffic issues depending on the time of day causing the program to hang. I will give another shot during off peak hours...
Happily making circuits for fun and entertainment...

KjellMorgenstern, 2021-07-06:
Solution: 0.9.7 will not pull github for updates.

closed 2021-07-25 by KjellMorgenstern