fritzing/fritzing-app #3837: Preferences not enabled

SciorraR, 2021-06-06:
## Current Behaviour
Text
Version 0.9.6 Preferences menu is grayed out.

**Operating System:**
<Mac version 11.4>
[image]
[image]

KjellMorgenstern, 2021-06-06:
Please use the official build from fritzing.org/download.

SciorraR, 2021-06-07:
Definitely a bug, I have the latest build

KjellMorgenstern, 2021-06-08:
I retested this after updating to macOs 11.4, still can not reproduce.

Did you upgrade from a previous version of Fritzing? If yes, which one? Maybe some previous files interfere.

Fabri54, 2021-07-08:
I've exactly the same issue. Latest release of Big Sur and latest, official, release of Fritzing.
Has anybody found a fix yet? Where should be the preference file?

failiz, 2021-07-08:
The same error has been reported in #3829, always on MacOS.

KjellMorgenstern, 2021-09-24:
Seemed fixed during tests, but this issue is happening quite random, so we can't close it yet.

sanmaneric, 2022-03-10:
This issue persists on Monterey 12.2.1 on Apple Silicon, macOS build 21D62. Fritzing version 0.9.9 downloaded from the official site.

failiz, 2022-03-10:
I think Kjell fixed it a few weeks ago. It will be fixed in Fritzing 0.9.10, which should be released soon.

makermelissa, 2022-05-20:
I'm still experiencing this with the latest code. As a workaround, it seems that if you go to the Code tab, it is enabled, but disabled on all other tabs.

KjellMorgenstern, 2022-05-20:
We still observed it on 0.9.10 during testing. It seems only to happen on first start, when the config file needs to be created or migrated. So simply restarting could help. I didn't check in detail, since finally getting out 0.9.10 has priority.

failiz, 2023-09-14:
Does this still happen in 1.0.1? If so, the workaround should be stated in the known problems of the release.

KjellMorgenstern, 2025-08-07:
I think this is fixed since the 1.0.0 release, now new reports since then.