fritzing/fritzing-app #3842: I have just installed frizzing and I am unable to edit the component properties in the inspector panel the component values are grayed out

pnaybour, 2021-06-12:
## Current Behaviour
I have just installed frizzing and I am unable to edit the component properties in the inspector panel the component values are greyed even on the example stretches.  
[image]

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
Download and install ftirzing on windows 10
Open example
Select a resistor from the schematic
Component values are set at 220ohmes and changes are greyed out. 

## Expected Behaviour
To be able to change the values of components such as resitors.

KjellMorgenstern, 2021-06-12:
The screenshot shows that the opened file is read only.

pnaybour, 2021-06-12:
A reboot fixed it. Sorry.

closed 2021-06-14 by KjellMorgenstern