fritzing/fritzing-app #3844: Can not create a double row connector on the breadboard view

evantill, 2021-06-15:
## Current Behaviour
Changing a generic female Header from single row to double rows will not update the bread board view :

- PCB view is updated with a double row connector.
- Schematic view is displaying a single row connector.
- Breadboard view is displaying a single row connector.

**Build:**
Version 0.9.6 (bCD-178-0-8a1e0682 2021-02-21) Cocoa [Qt 5.12.2]

**Operating System:**
macOS Mojave 10.14.6

**Steps to reproduce:**
- Create a new schematic
- Add a generic female Header
- Change pin numbers to 8 Pins
- Change row to **Double**

## Expected Behaviour
Bread board view and schematic view display a double row connector.

evantill, 2021-06-15:
Back in 2015 https://github.com/fritzing/fritzing-app/issues/3085#issuecomment-152519962 has reported the same problem

vanepp, 2021-06-23:
This is working as designed (if not as desired!) I was told back in 2016 that the thought was that dual row headers couldn't be properly represented in breadboard so the parts factory doesn't try and do so. That is obviously incorrect since you can create one from two single row headers just fine but the parts factory has never been updated. This would be an enhancement not a bug.

evantill, 2021-06-28:
>   I was told back in 2016 that the thought was that dual row headers couldn't be properly represented in breadboard
Maybe not on a breadboard but I was thinking to use it on a perfboard or on a stripboard ( where it can be properly represented).

KjellMorgenstern, 2021-06-28:
We want to improve support for Perfboard / Stripboards. The behavior should be adapted.

vanepp, 2021-06-30:
There is no reason I know of that the parts factory can not emit double row connectors in breadboard, it just does not do so now because it was considered not a good thing to do in 2016. I think we should change parts factory to emit double row connectors. There are a bunch of changes I'd like to make in parts factory but there is code to support previous versions of connectors which make making changes difficult (at least for me!)