fritzing/fritzing-app #3850: Mouse click on the parts search does not start the search only the return key.

vanepp, 2021-06-24:
## Problem

When doing ng a parts search mouse clicking on the search icon will not start the search, only hitting the return key will start the search. 

## Proposed Solution
Make a mouse click on the search icon act the same as the enter key. So a mouse click on the search icon (circled in red) will start the search. 

[image]

KjellMorgenstern, 2021-07-30:
[image]

[image]

Proposed improvement:
1. Search updates while typing
2. Replaced search icon with a clear button

Currently, rendering the icons for the search result is what slows the search. There are two possible solutions.
a. update the icons with a thread pool, in the background
b. store the icons as blobs in sqlite, which should load them almost instant.

Generating the icons is burried quite some levels deep, the same functionality also delays Fritzing when starting up.

KjellMorgenstern, 2022-02-24:
The searching glass issue has been resolved by replacing it with a "x" button for clearing the search for now.
We are working on further improvements to the search function.

closed 2022-06-01 by KjellMorgenstern