fritzing/fritzing-app #3851: Make dragging past the end of the window scroll the image.

vanepp, 2021-06-24:
## Problem
Starting from this configuration, drag the IC up 

[image]

in this instance Fritzing works as desired, the image scrolls up as long as the mouse is near the edge of the screen and the board outline moves down in the image. 

[image]

however doing the same to the left gets no movement and a circle with a line through it (which I haven't been able to capture with screen capture software) and the screen doesn't scroll as it did in the up direction

[image]

here I am trying to scroll left, but instead of scrolling I get the circle with a line through it just past the edge of the screen on the left. Late discovery, it looks like moving the mouse cursor past the edge of the screen stops the scrolling and gives the circle with the line through it there after til something resets. Changing to the other direction then works again (i.e. changing from scrolling down to scrolling up works as long as the mouse cursor doesn't go above the active part of the screen!) Going past the active part of the screen appears to set something that blocks scrolling! In schematic view the scrolling is limited. The scroll works for about one and a half  0.1 grid lines then the part moves back to its original position and scrolls .15in again in a loop. Breadboard works correctly until the mouse cursor moves beyond the edge of the active screen then the circle with a line through it is the response from then on until you scroll the other way or release the mouse and reselect the part.

## Proposed Solution

Dragging an object off the screen in any direction in any view will cause the image to scroll (Inkscape does this, but I don't think Inkscape uses QT!) It looks like QT can do it if we figure out what is tripping the stop. action.

KjellMorgenstern, 2021-07-30:
Let's limit the issue to the PCB view for now.

Confirmed with 0.9.7:
Scrolling works in each direction
However, one has to carefully move the mouse to the window boarder, where the scrolling starts.
As soon as the mouse moves a bit to far, the scrolling operation stops, and will not continue anymore.
Only way to scroll again is to drop the part and pick it up again, starting a new scroll operation when moving to the window border.

vanepp, 2022-07-15:
It looks like the fix in  #3959 didn't entirely fix this issue. It appears the problem occurred between 0.9.8 and 0.9.9.

A test sketch (remove the trailing .zip to get the .fzz file)

[scroll-test.fzz.zip](https://github.com/fritzing/fritzing-app/files/9124218/scroll-test.fzz.zip)

on 0.9.8 load the sketch and change to pcb view and scroll to the bottom of the screen

[image]

drag R2 to the bottom edge of the screen, scroll size increases as desired.

[image]

Load the test file on 0.9.9

load the .fzz, change to pcb view, scroll window to bottom as with 0.9.8

[image]

drag R2 to the bottom edge of the screen, screen size does not increase any more. Moving the resistor down will move it out of the window, but not increase the scroll length and you can no longer get to R2 unless you scroll out

[image]

I can move R2 out of the window, but then I can no longer access it without scrolling as the window won't scroll down

[image]

R2 is below the edge of the screen here but I can't scroll down to access it. Zooming out allows me to see R2 again.

[image]

zooming in makes R2 inaccessible again

[image]

so the problem looks to have occured between versions 0.9.8 and 0.9.9 and is not completely fixed in Version 0.9.10 (bCD-1801-0-29c2cede 2022-05-14) 64 [Qt 5.15.2] the latest release with a change backed off to fix this bug, doesn't quite fix it entirely.

[image]

[image]

KjellMorgenstern, 2022-07-16:
@vanepp This might be a bug that is a bit subtle, so observations can differ from screen size, OS and Fritzing version, to the point that it is very difficult to tell at which time it was introduced.

Can you confirm this observation:

When you drag a part to the edge, to trigger scrolling, be very careful to stay close to the edge, about 5-10 pixels outside of the sketch area. This way, scrolling works reliably in each direction.
*But*
When you exceed a certain distance from the sketch area, the mouse pointer turns into a :no_entry_sign: or similar. From then on, no scrolling at all is possible, until you release the tile somewhere on the sketch area and start a new drag operation.

I believe this behavior is in Fritzing since a long time.

KjellMorgenstern, 2022-07-16:
There is a relation with #3959, but they are not the same bugs.
In the 0.9.10 release, we tried to get a more reliable and reproducible behavior of the "fit screen" <kbd>Ctrl</kbd><kbd>0</kbd> feature. This is important for our automated tests, but also for user experience (it can be quite confusing if centering the project on the screen doesn't work right).
However, by recalculating the sketch area, we probably narrowed the trigger area for scrolling, which caused this issue (#3851) to happen almost all the time.

vanepp, 2022-07-17:
"However, by recalculating the sketch area, we probably narrowed the trigger area for scrolling, which caused this issue (https://github.com/fritzing/fritzing-app/issues/3851) to happen almost all the time."

I think this is the correct description of the problem. as of 0.9.10 the problem happens all the time in my experience, but it is possible that scrolling will still work if you get close enough to the edge of the screen. I just tried again on 0.9.10 (meaning it is probably true on 0.9.9 as well) and on my first attempt indeed I managed to scroll as usual, but three attempts after that (unselecting and the reselecting the part again each time) give the circle with the line through it and won't scroll. While you are there can the trigger area be made a little wider reasonably easily? Having scrolling work reliably would be very desirable in my opinion!