fritzing/fritzing-app #3853: App crashes when a second (non-primary) Monitor is to the left side

YannikSc, 2021-06-26:
## Current Behaviour

When I try to **launch** Fritzing on my primary monitor while I have a secondary monitor configured on the left side, the application crashes

<details>
<summary>Debug console output</summary>
```
"candidate:/home/yannik/.local/share/fritzing"
"candidate:/usr/local/share/fritzing"
"candidate:/usr/share/fritzing"
"data folders found: /usr/share/fritzing"
"translation en_US loaded 1 from /usr/share/fritzing/translations"
Icon theme "adwaita" not found.
Icon theme "ubuntu-mono" not found.
Icon theme "yaru" not found.
Icon theme "Mint-X" not found.
Icon theme "elementary" not found.
"registering font family: Droid Sans DroidSans"
"registering font family: OCRA OCRA"
"got transaction"
"referenceModel::loadAll completed"
"parts count 2155"
"load something"
"checking for backups"
"check files to load 0"
"create empty sketch"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
"loading bin 'Core Parts'"
"module id mystery_part_sip_3_100mil not found in database"
"module id generic_ic_dip_8_300mil not found in database"
"module id generic_female_pin_header_2_100mil not found in database"
"module id generic_shrouded_pin_header_10_100mil not found in database"
"module Mystery Part - 3 pins:mystery_part_sip_3_100mil obsolete svg /usr/share/fritzing/fritzing-parts/svg/obsolete/icon/mystery_part_icon.svg"
"module Generic female header - 2 pins:generic_female_pin_header_2_100mil obsolete svg /usr/share/fritzing/fritzing-parts/svg/obsolete/icon/generic_female_header.svg"
"Setting platform to Arduino"
"updating link"
"Setting board to Arduino BT"
"Setting board to Arduino UNO"
"Setting board to Arduino UNO"
"Setting board to Arduino UNO"
"Setting port to ttyACM0"
[1]    405042 segmentation fault (core dumped)  Fritzing --debu
```
</details>

When I change the monitor to the right side, or mark the left monitor as primary, the application launches successfully.
It also works when I move the desktop icon to the left monitor and launch the application there.

**Build:**
From "About Fritzing": Version 0.9.6 (b ) 64 [Qt 5.15.2]

**Operating System:**
Garuda Linux with KDE
plasmashell 5.22.1
Kernel: Linux 5.12.10-zen1-1-zen #1 ZEN SMP PREEMPT Thu, 10 Jun 2021 16:34:48 +0000 x86_64 GNU/Linux

**Steps to reproduce:**
- Connect a second monitor
- Mark the right one as primary in the system settings
- Launch Fritzing on the right monitor -> SegFault
- Launch Fritzing on the left monitor (e.g by moving over the terminal or desktop icon) -> Works

## Expected Behaviour
- Fritzing starts no matter how the monitor layout is and on which monitor it is launched

KjellMorgenstern, 2021-06-26:
Which one is the external monitor? Are you on a notebook, do one or both of the screens have retina ( for example > 250dpi) resolution?

KjellMorgenstern, 2021-06-26:
Please add full version info from "About Fritzing"

YannikSc, 2021-06-27:
I'm on a desktop, 1080p secondary, left monitor and a 4k as primary on the right side. Both running an 100% resolution. 
The provided version is already all I got. I installed it on arch through the AUR, maybe this interferes?

KjellMorgenstern, 2021-07-30:
I tried 0.9.6 and 0.9.7 with various screen configurations on Ubuntu 20.04. Both had the same result:

Pimary screen  | Fritzing start | Outcome
---|---|---
right | right | Fritzing starts on primary screen
right | left | Fritzing starts on primary screen
left | right | Fritzing starts on primary screen
left | left | Fritzing starts on primary screen

janclod, 2021-08-20:
I also tried launching the 0.9.6 on Ubuntu 20.04 and it seems to work.

I have to add that I am launching from the command line, could this make a difference?