fritzing/fritzing-app #3855: Mouse disappears over breadboard / components in Wayland session

benstockil, 2021-07-07:
## Current Behaviour

The mouse cursor disappears when hovering over the breadboard, or over any component in schematic or PCB views. 

[image]

**Build:** 
Version 0.9.6 (b) 64 [Qt 5.15.2] _(Installed from AUR)_

**Operating System:**
Arch Linux
Wayland / Sway (v. 1.6)

**Steps to reproduce:**
Hover over breadboard / component with mouse (see GIF)

benstockil, 2021-07-07:
I should also mention that this only happens when running under native Wayland; this bug goes away when running under XWayland (by running `env -u WAYLAND_DISPLAY Fritzing`).

KjellMorgenstern, 2021-10-05:
Can not reproduce. Probably this is a third party build.

closed 2021-10-05 by KjellMorgenstern

roberti42, 2022-01-11:
I also experience this bug when running on Fedora 35 (and 34) on sway.

KjellMorgenstern, 2022-01-11:
Please add exact version information, including hash and build number, otherwise it does not make sense to reopen the issue.

roberti42, 2022-01-18:
These are the only information I found in Help > About:

Version 0.9.6 
(b ) 64 [Qt 5.15.2]

@KjellMorgenstern how to find the hash and build number?

whitelynx, 2022-05-10:
I've also been experiencing this issue consistently for a couple of years now (since sometime in 2019) on Wayland/Sway under Manjaro Linux.

Here's my current build:
Version 0.9.9 
(b0.9.6-35-gf0af53a9 2021-09-22) 64 [Qt 5.15.3]

It is a local build, installed using [the AUR package for Fritzing](https://aur.archlinux.org/packages/fritzing).

KjellMorgenstern, 2022-05-11:
Did you try our official build from fritzing.org ?

whitelynx, 2022-05-19:
I paid to download the official build from fritzing.org just now, and it's even worse - it just gets a `SIGABRT` at startup:
```
Reading symbols from /home/whitelynx/bin/fritzing-0.9.9-l348-f0af53a9.AppImage...
(No debugging symbols found in /home/whitelynx/bin/fritzing-0.9.9-l348-f0af53a9.AppImage)
(gdb) r
Starting program: /home/whitelynx/bin/fritzing-0.9.9-l348-f0af53a9.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 150499]
process 150357 is executing new program: /usr/bin/env
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
process 150357 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 150504]
process 150357 is executing new program: /tmp/.mount_fritzi5uEo2J/usr/bin/Fritzing
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland-egl" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Program received signal SIGABRT, Aborted.
0x00007ffff3e5f34c in __pthread_kill_implementation () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff3e5f34c in __pthread_kill_implementation () from /usr/lib/libc.so.6
#1  0x00007ffff3e124b8 in raise () from /usr/lib/libc.so.6
#2  0x00007ffff3dfc534 in abort () from /usr/lib/libc.so.6
#3  0x00007ffff42929bc in QMessageLogger::fatal(char const*, ...) const () from /tmp/.mount_fritzi5uEo2J/usr/bin/../lib/libQt5Core.so.5
#4  0x00007ffff5b67fa0 in QGuiApplicationPrivate::createPlatformIntegration() () from /tmp/.mount_fritzi5uEo2J/usr/bin/../lib/libQt5Gui.so.5
#5  0x00007ffff5b682ad in QGuiApplicationPrivate::createEventDispatcher() () from /tmp/.mount_fritzi5uEo2J/usr/bin/../lib/libQt5Gui.so.5
#6  0x00007ffff44a4fe3 in QCoreApplicationPrivate::init() () from /tmp/.mount_fritzi5uEo2J/usr/bin/../lib/libQt5Core.so.5
#7  0x00007ffff5b6accc in QGuiApplicationPrivate::init() () from /tmp/.mount_fritzi5uEo2J/usr/bin/../lib/libQt5Gui.so.5
#8  0x00007ffff6567aa9 in QApplicationPrivate::init() () from /tmp/.mount_fritzi5uEo2J/usr/bin/../lib/libQt5Widgets.so.5
#9  0x00005555554fb6a1 in ?? ()
#10 0x00005555554b26dd in ?? ()
#11 0x00007ffff3dfd310 in __libc_start_call_main () from /usr/lib/libc.so.6
#12 0x00007ffff3dfd3c1 in __libc_start_main_impl () from /usr/lib/libc.so.6
#13 0x00005555554e11ca in ?? ()
(gdb) 
```

KjellMorgenstern, 2022-05-19:
`env - fritzing.AppImage` is not supposed to work. For example, try `env - firefox`.
The error message is similar, but I admit the firefox error message is clear.

Is XDG_SESSION_TYPE=wayland a default on your system? 
Your approach to unset the variable makes sense, but you should not unset *all* of them. DISPLAY is needed after all.

I would try `XDG_SESSION_TYPE="" ~/bin/fritzing-0.9.9-l348-f0af53a9.AppImage`

whitelynx, 2022-05-19:
Yes, `XDG_SESSION_TYPE=wayland` is a default on my system. I'm still not sure why it would crash that way, since the one I installed through my package manager works fine using Wayland itself other than the cursor issue mentioned in this issue.

Unsetting only `XDG_SESSION_TYPE` still crashes:
```
 env XDG_SESSION_TYPE='' ~/bin/fritzing-0.9.9-l348-f0af53a9.AppImage
qt.qpa.plugin: Could not find the Qt platform plugin "wayland-egl" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

fish: Job 1, 'env XDG_SESSION_TYPE='' ~/bin/f…' terminated by signal SIGABRT (Abort)
```

Even setting it to `x11` explicitly crashes:
```
 env XDG_SESSION_TYPE='x11' ~/bin/fritzing-0.9.9-l348-f0af53a9.AppImage
qt.qpa.plugin: Could not find the Qt platform plugin "wayland-egl" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

fish: Job 1, 'env XDG_SESSION_TYPE='x11' ~/bi…' terminated by signal SIGABRT (Abort)
```

whitelynx, 2022-05-19:
Aah, didn't realize I had that set. It does run when that is also unset.

Still, I think it would make sense to see if we can get the custom cursors working under native Wayland as well. When running under XWayland, the cursors appear but the hotspot is almost an entire cursor-length away from where it should be.

T-vK, 2024-08-09:
For our fellow `flatpak` users:
``` Bash
flatpak run --env=QT_QPA_PLATFORM=xcb org.fritzing.Fritzing
```

tronta, 2024-12-31:
Can be also added via flatseal
[image]

Schneckensuppe, 2025-08-11:
solve for my debian 12, gnome 43.9 
launching from the terminal:
`XDG_SESSION_TYPE=x11 fritzing`