fritzing/fritzing-app #3857: 180 degree rotation causes seg fault in geber export.

vanepp, 2021-07-22:
## Current Behaviour

this sketch with a single smd transistor from core parts rotated 180 degrees causes a seg fault in gerber processing when exported. Changing the rotation to 179 degrees works (and is the work around.) See 

https://forum.fritzing.org/t/fritzing-0-96-version-crashes-when-exporting-gerber-file/13467/

for further details if needed. 

[relay-led-motor-V2_2_gnd_2switch_V3.fzz.zip](https://github.com/fritzing/fritzing-app/files/6864547/relay-led-motor-V2_2_gnd_2switch_V3.fzz.zip)

remove the .zip to get the sketch. This will crash when exported as a gerber. Changing the rotation to 179 degrees exports correctly. 

[image]

**Build:**

Version 0.9.6 (bCD-175-0-8a1e0682 2021-02-21) 64 [Qt 5.12.10]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Step 1 export fzz file as gerber crashes. 
- Step 2 change rotation from 180 to 179 export gerber works correctly. 

## Expected Behaviour

gerber exports without seg faulting.

KjellMorgenstern, 2021-07-30:
I can reproduce this in 0.9.6.
In 0.9.7 the export works fine.

I get the warnings
"Copper0 layer empty"
"exported mask layer Mask0 is empty"
"exported paste mask layer is empty"
which is expected, since there are no copper elements on the bottom layer.

vanepp, 2021-07-31:
I agree, the original file now renders correctly on 0.9.7 so either the bug was fixed by a change in 0.9.7 or has moved to a new value we haven't found yet. In either case we can close this issue and report it again if we find a new instance.

closed 2021-07-31 by vanepp