fritzing/fritzing-app #3859: Generic IC broken on 0.9.7

vanepp, 2021-07-30:
## Current Behaviour

Upgraded to 0.9.7 from 0.9.6 (without removing 0.9.6 first)

Drag a generic IC in to the sketch and try and resize it. 

[image]

resize fails

[image]

instead of the expected 10 pin dip, the 8 pin remains (in all views) 

[image]

trying to change to an so package also fails, package remains 8pin dip. 

[image]

Import a generic 20 pin dip created with 0.9.3b probably. Attempting to modify it also fails. 

[image]

switch to schematic and try editing pin labels set new tab editable pin labels from false to true to enable pin labels (presumably!) 

[image]

works (but doesn't allow you to go back to non editable pin labels!) but changes the IC from 20pins to 8

[image]

running back the undo may indicate the problem. Here it is going to undo Swapped IC2 (which was a 20 pin dip) with generic 8 pin DIP which is likely the problem

[image]

which indeed restores the 20 pin dip as expected. 

[image]

I also tried uninstalling Fritzing 0.9.7, rebooted the machine and then reinstalled 0.9.7 (without having the previous copy of 0.9.6 in the mix) but there was no change so it doesn't appear to be an upgrade related issue.

In addition, trying to change the 20pin to a 10 pin (which used to be legal) gets this message;

[image]

which seems to also be incorrect. 

**Build:**

Version 0.9.7 
(bCD-534-0-120da230 2021-07-12) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Step 1 see current behaviour
- Step 2

## Expected Behaviour
Expected to be able to modify the generic IC as in previous versions.

KjellMorgenstern, 2021-07-30:
Confirmed.
Is using the mystery part ("?") instead of the generic ic ("IC") an alternative?

vanepp, 2021-07-31:
"Confirmed.
Is using the mystery part ("?") instead of the generic ic ("IC") an alternative?"

The mystery part will do some of it, but schematic (mystery part on the left)is single side not double.

[image]

so the result is quite different. I think the best thing would be to move the fix in to a replaced 0.9.7, otherwise the likely workaround will be to down grade to 0.9.6 where it works (this is what the original poster in the forums ended up doing). The generic IC is a quite important part!

vanepp, 2021-08-09:
Fixed in 0.9.8, thanks Kjell!

closed 2021-08-09 by vanepp