fritzing/fritzing-app #3860: Can not open file

mmgasparovic, 2021-08-03:
## Current Behaviour
Can not open (load)
[Asus N552VW 60NB0AN0-MB1C00.zip](https://github.com/fritzing/fritzing-app/files/6923332/Asus.N552VW.60NB0AN0-MB1C00.zip)
 file.
[image]
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Open file in attachment.

## Expected Behaviour

KjellMorgenstern, 2021-08-03:
I unpacked the zip file to Asus\ N552VW\ 60NB0AN0-MB1C00.fz 
It should be xml data, but it is not.
It does not have a recognizable file header.
Shannon Entropy of the files is 7.9991 . This is near the theoretical maximum (8), which is a hint that it is random data or strong encryption.

KjellMorgenstern, 2021-08-22:
Closing as we can't do much here.

closed 2021-08-22 by KjellMorgenstern