fritzing/fritzing-app #3861: Use LunaSVG for importing SVGs

sammycage, 2021-08-06:
## Problem
> Describes from a user perspective the current state and its shortcomings.

QtSVG is not maintained and is not as robust #3805 #3589

## Proposed Solution
> Describes from a user perspective the ideal state of the implementation

[LunaSVG](https://github.com/sammycage/lunasvg) is a standalone SVG rendering library in C++. It's fast, easy to use, regularly maintained, and supports a wide range of svg elements. [PICsimLab](https://github.com/lcgamboa/picsimlab) uses [LunaSVG](https://github.com/sammycage/lunasvg) to import PCBs created by [PcbDraw](https://github.com/yaqwsx/PcbDraw).

Basic Usuage :

```cpp

#include <lunasvg.h>

using namespace lunasvg;

int main()
{
    auto document = Document::loadFromFile("board.svg");
    auto bitmap = document->renderToBitmap([<max width>, <max height>, <background color>]);

    // do something useful with the bitmap here.

    return 0;
}
```

KjellMorgenstern, 2021-08-06:
Fritzing does not only need to render SVG, but also must parse and transform them into an internal representation of the electric circuit. It also generates gerber files from SVGs and re-renders multiple SVGs into one file.
Most issues raise from this fact. Incomplete SVG 2.0 support (animations, full CSS) which results from using QtSVG is less of an issue.

Replacing the parser was considered in early 2020, but we found at that time that it is not feasible.
In the LunaSVG docs it says that Text elements are not supported, were I would currently stop evaluating.

Of course it would be interesting to render some of the Fritzing graphics with different tools for a first comparison. Note, such a render test will not yet cover Fritzings mechanisms to re-render into gerber and other formats. 

For a more generic approach, it might make sense to run LunaSVG against a more official test set,
for example

https://razrfalcon.github.io/resvg-test-suite/svg-support-table.html
or
https://www.w3.org/Graphics/SVG/Test/20110816/

closed 2021-08-06 by sammycage

failiz, 2021-08-06:
FYI, I just found this while browsing hackaday. It seems a bit related. Maybe it is interesting.

https://github.com/jaseg/gerbolyze