fritzing/fritzing-app #3862: Unable to modify generic IC on MAC OS

MikeTinsley, 2021-08-09:
## Current Behaviour
Attempting to change number of pins on generic IC from 8 to 16 (or anything else) results in this error:

 Parse error (2) at line 1, column 1:
unexpected end of file
/Users/michaeltinsley/.config/Fritzing/partfactory/eda1b1e76ca3627c596ba0a2b34794ab/core/generic_ic_dip__300mil.fzp
[image]

**Build:**
Version 0.9.7
**Operating System:**
macOS Big Sur 11.4

**Steps to reproduce:**
select generic IC and place on schematic
attempt to change number of pins

## Expected Behaviour
IC is modified accordingly

KjellMorgenstern, 2021-08-09:
Duplicate of #3859

closed 2021-08-09 by KjellMorgenstern

MikeTinsley, 2021-08-09:
Have now downloaded 0.9.8 and it works - problem solved