fritzing/fritzing-app #3863: Fritzing crashes when swapping a superflux led

failiz, 2021-08-10:
## Current Behaviour
1. Add a superflux Blue Led in the breadboard like in the picture:
[image]

2. Change the property polarrity to "common cathode"
3. Fritzing crashes
 
<Windows 10>, Fritzing 0.9.8

KjellMorgenstern, 2021-08-11:
Could reproduce this also with Fritzing 0.9.6 on Ubuntu 20.04

failiz, 2022-01-12:
Not sure if it is the same bug, could you check this in the development branch:
Add an Arduino Nano (3.0)  to the breadboard view
Change it to Arduino Nano (3.0) with ICSP 
Rotate it 90 degrees
Place it on the breadboard
Change it to Arduino Mini (Rev5)
Fritzing crashes in 0.9.9...

closed 2022-06-01 by KjellMorgenstern