fritzing/fritzing-app #3868: Geber processing in 0.9.8 still has close path issues.

vanepp, 2021-08-19:
## Current Behaviour

While the path used is a little odd (the cutouts come first followed by the outline) it renders correctly in Inkscape and in pcb view in Fritzing but the left closing line is missing in the gerber output. 

[camdenboss-3a-screw-old.svg.zip](https://github.com/fritzing/fritzing-app/files/7011188/camdenboss-3a-screw-old.svg.zip)

(remove the trailing .zip to get the svg file!)

file renders correctly in Fritzing pcb view

[image]

but is missing the closing line in the gerber output, then load the svg on to Fritzing via the load image file button in pcb view. 

[image]

This svg (where the trailing Z has had a v-360 added to the path to bring it back to the start works correctly. 

[camdenboss-3a-screw.svg.zip](https://github.com/fritzing/fritzing-app/files/7011265/camdenboss-3a-screw.svg.zip)

(remove the trailing .zip to get the svg file!) and is the work around that is being used. 

**Build:**

Version 0.9.8 (bCD-92-0-b3d4fefa 2021-08-03) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Load camdenboss-3a-screw-old.svg in to pcb view via load image file in Inspector.
- File->Export->For Production->Extended Gerber 
- View the Untitled Sketch_contour.gm1 file with a gerber viewer.
- The left (closing) line of the board outline is not rendered. 

## Expected Behaviour
The complete rectangle for the outline should be rendered.

KjellMorgenstern, 2021-08-19:
Possibly the same as #3820

KjellMorgenstern, 2021-08-22:
[image]

closed 2022-06-01 by KjellMorgenstern