fritzing/fritzing-app #3869: SIGSEGV crash

UltraBlackLinux, 2021-08-20:
## Current Behaviour
Fritzing immediately crashes when I am trying to run it: 
```
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
fish: Job 1, 'Fritzing' terminated by signal SIGSEGV (Address boundary error)
```

**Build:**
0.9.6-1 [+228 ~0.88] (???)
Installed from the AUR

**Operating System:**
Garuda Linux

KjellMorgenstern, 2021-08-20:
Can not reproduce, the official 0.9.6 build runs fine.
Did you try our official (paid) 0.9.8 AppImage build?

closed 2021-08-20 by KjellMorgenstern

UltraBlackLinux, 2021-08-20:
> Can not reproduce, the official 0.9.6 build runs fine.
> Did you try our official (paid) 0.9.8 AppImage build?

I don't have the money to pay 8$ for trying software once, that might not suit my needs, so I don't. (I hope that didn't sound too rude, I just wanted to prove my point)

UltraBlackLinux, 2021-08-20:
Sadly https://github.com/AppImage/pkg2appimage/blob/master/recipes/Fritzing.yml also doesn't work anymore as every (obvious) way of omitting the paywall has been removed

KjellMorgenstern, 2021-08-20:
No problem, I understand your point. But without the payment, there wouldn't be a version 0.9.6 or 0.9.8, and probably not even a server reachable on fritzing.org ...

UltraBlackLinux, 2021-08-20:
> No problem, I understand your point. But without the payment, there wouldn't be a version 0.9.6 or 0.9.8, and probably not even a server reachable on fritzing.org ...

I have been searching for a while and I can't find any fritzing alternatives, that have the support and are open source.
Would you maybe send me the newest appimage build via mail or something, so that I can try, if It's crashing too? I will buy the appimage build, when I can be sure, that it will not crash, otherwise it's a waste of money.
I hope it's possible.

UltraBlackLinux, 2021-08-20:
Anyways, after a lot more trying random stuff out, I found something out: The app "launches" when the fritzing parts directory is missing or renamed, but errors a ton.
After some more trying it turns out, that fritzing is crashing when there is a folder in the main directory called fritzing-parts, but it's all fine and loading when the folder is called "parts"
I don't get this