fritzing/fritzing-app #3875: Fritzing is extremely slow and glitch on stripboard resize

georgyfarniev, 2021-08-29:
## Current Behaviour
Text
Fritzing gets unusably slow on stripboard resize. I observe [this bug](https://github.com/fritzing/fritzing-app/issues/2998) and also fritzing becomes extremely slow and unusable. As a person who worked with **QGraphicsScene** before, I can say that it's abnormal behavior. Everything else works fine, intel gpu drivers are up to date.
**Build:**
Version 0.9.8  (bCD-92-0-b3d4fefa 2021-08-03) 64 [Qt 5.15.2]

**Operating System:**
Windows 10
16GB RAM, Intel i5-7440HQ, No Discrete GPU

**Steps to reproduce:**
- Create new project
- Delete a breadboard and add a stripboard from parts section
- Resize stripboard by adding rows, let's say, change 20 to 54

## Expected Behaviour
Stripboard resized correctly without issues

KjellMorgenstern, 2021-08-29:
Duplicate of #3084

failiz, 2021-09-25:
#3084 has been fixed in 0.9.9 and the issue is closed now. This should be closed too.

closed 2021-09-25 by KjellMorgenstern