fritzing/fritzing-app #3876: github repository out of date

michaellass, 2021-08-31:
It looks like Fritzing's github repository is not updated anymore. Since version 0.9.6 (8a1e0682d7846f6384ae45373961ed693b4c55d2) there are only five commits on the develop branch, of which only one addresses bugs from this issue tracker (1f0953f4f5a6137a667ee5a1e5b87a916f925e5b). The changelog at https://fritzing.org/releases/0-9-8 and the blog entry at https://blog.fritzing.org/2021/08/09/fritzing-0-9-8 mention 250 bugs from this tracker being fixed between 0.9.6 and 0.9.8. This doesn't really fit together.

Am I missing something or is the source code for Fritzing 0.9.8 currently not available? If so, is this an oversight or an intentional change?

altimmons, 2021-09-02:
I am wondering the same.  It might have to do with the enforced donation scheme.

I think if its going to transition from Open to Closed source- which it appears to have, it needs to be renamed for clarity.
Call it fritzing Plus or something,

lx1, 2021-09-04:
> I think if its going to transition from Open to Closed source- which it appears to have, it needs to be renamed for clarity.
> Call it fritzing Plus or something,

It would be a pretty common solution for cases like this. But it would require an initial distinction between the free version and the non-free one(s), with an explicit permission for the free code to be used in the non-free version. Instead, Fritzing's particular license statement contains something that explicitly prevents such a thing from happening:
_The source code of Fritzing is under GNU GPL v3 ... This means that you can create your own variation of Fritzing, as long as you credit us and also publish it under GPL._
I think it's clear the initial intent to create and keep Fritzing as a free software initiative. This is fundamental that anybody who uses or anyhow relate to fritzing has to take into account.

altimmons, 2021-09-05:
True and I agree, but it can no longer claim to be free as the last 3 updates are not available as source.  

It’s unusual. I strongly support open source, but fritzing did die for a bit.  There is a lot more I would like to see it do, and if someone is actively developing it - full time then perhaps.

in this case there should be an advisory committee of invested people- who can take in the proceeds and decide on Human Resources. Other wise it’s a bit odd

lx1, 2021-09-06:
Hi altimmons. If "the last 3 updates are not available as source", it's a simply a violation of the license.  It's just strange that this comes from the current development team and not from a concurrent third party stealing the code and using it in their proprietary product. This from a license point of view.

Then I clearly understand that Fritzing was almost starving.  I think that if the current developers would fork the source code, call it somewhat else (maybe Paying? :) , making it clear what operation is being done, I think that nobody would disagree, even it from a legal point of view, this would be another GPL violation. But it has to be clearly stated, so people can choose between pying and having the new proprietary product in zero time, or forking Fritzing and continuing to evolve it. This is what for exmple happened to the popular MySQL database engine, and to numerous other software. Obviously MySQL was originally licensed in a much looser way than GPL, so the owner of it could continue to call it MySQL and the frre forks which were born from the last open version had to be renamed.

KjellMorgenstern, 2021-09-06:
@michaellass: Yes, you are right. It was partly an oversight, and I will resolve this. But it also hints at the intention/direction: Not necessarily closed source, but certainly a model that allows to pay people for their work long term. E.g., without others distributing 'free' copies against a few bucks from advertisements (not saying that this is with bad intent, but it doesn't really help). Fritzing needs more than software developers. If that were the issue, I'd spend most of my time writing code and fixing bugs. But as I learn from this thread, the current situation is still confusing, and we need to improve public communication. Any PR people in Berlin interested? Resolving the deadlock that led to the zombie state from 2016-2019 is still a significant task. And while reducing the GitHub backlog (1106 down to 546 right now), we are starting to get in new features. It is pretty complex, but it is worth it. Otherwise, I wouldn't work on this full time since more than two years and even hire people. 

@lx1: Is this meant as advice? If your rights are violated, I suggest to contact me directly in time.  I hope you understand I can't help much with anonymous comments on GitHub.

lx1, 2021-09-06:
> @lx1: Is this meant as advice? If your rights are violated, I suggest to contact me directly in time. I hope you understand I can't help much with anonymous comments on GitHub.

Hi Kjell. I'm LuxT from the forum, I think we just met about this particular question. I'm just trying to understand how the current situation relates to GPL, being that Fritzing is under GPL. I'm trying to understand if we're only going through a period of time where the source on Github is outdated (i.e. an oversight) or if there is some precise plan to not release the source any more. If it was the latter case, yes, I would think that my (and anyone') rights would be violated, since Fritzing _is_ under GPL, and this is a fact we can't ignore, I think.

Regarding the advice, I just wrote what I would do, but this is not the only way, eh. Another possibility that comes to my mind: release the source on Github and use Github as the reference source, so GPL is not violated. After all, no one imposed to choose GPL at the origin: it was only a common choice for a project like this. Then, release the binaries only on a paid basis just as today, but clearly asking for no one to distribute concurrent binaries, which would soon hinder Fritzing development. I just think that all this confusion is NOT due to the fact that a fee is required to download the binaries. It IS due to the fact that the repo is not updeted, or seems to be not updated, since it still report 0.9.6. From what I have learnt in this thread and the forum thread I started, most people are interested in the binaries. They quite don't know of licenses at all, and are in no way interested in them. They just want Fritzing to work without bugs and with support for their components etc. This will give you the option to be paid for the "finished" version, without violating the license in any way, which is why all this mess began.

And more, I am one of the people who forked the source code as it is now, and distributed compiled 0.9.6 binaries on my site. This is absolutely not due to the desire to make a few bucks: my site does absolutely not contain banners, advertisements or so, and is not advertised anywhere. It was just a matter of principle, due to the fear that Fritzing was moving away from GPL, which in my opinion would be a license violation, yes. Now I'm happy that my initiative (and of others) raised the problem to the attention, and I can remove the post on my site. Regards, Luigi

reini-1, 2021-09-08:
So when will the latest source code be available?

KjellMorgenstern, 2021-09-30:
Sources are at 67355fc

closed 2021-09-30 by KjellMorgenstern

xerz-one, 2022-06-01:
Hey, just noticed there's a new version 0.9.10 for which the source code is not available! Can the source code please be updated?

lx1, 2022-06-07:
> Hey, just noticed there's a new version 0.9.10 for which the source code is not available! Can the Git repository please be updated? Thanks in advance!

Please take a look at the "main" branch, it contains numerous commits (it's somewhat confusing that the "develop" branch is behind the "main" bransh, but they are only names, after all). I did not notice it too, but in days I'll try to build it.

gsingh93, 2023-06-20:
The last commit on main or develop is from July 1st, 2022. Apparently 1.0.0 is now released: https://fritzing.org/releases/1-0-0

Where is the source code for it? Where is all of the development for this happening?

plazmax, 2023-09-10:
> The last commit on main or develop is from July 1st, 2022. Apparently 1.0.0 is now released: https://fritzing.org/releases/1-0-0
> 
> Where is the source code for it? Where is all of the development for this happening?

same as "1.0.1 - 06 September 2023"

https://fritzing.org/download/history-changes/

gsingh93, 2023-09-10:
In case anyone is looking for the official response for this: https://forum.fritzing.org/t/can-t-find-source-code/19723/7

failiz, 2023-09-10:
Source code for 1.0.0 is the development branch. I guess 1.0.1 source code will be committed soon.