fritzing/fritzing-app #3881: Source of 0.9.8?

hoppke, 2021-09-14:
Hi, where can I find the sources that were used to build 0.9.8, as offered on https://fritzing.org/releases/0-9-8 ?
Or is what fritzing.org is selling now a restricted/closed source fork of the public repo here?

michaellass, 2021-09-14:
Fritzing is selling their binary releases for quite a while already. Currently the corresponding source code is not available in this repository (see #3876). If you have bought the binary release, you should however have the right to obtain the corresponding source code according to the GPL.

hoppke, 2021-09-15:
Thanks. There are too many red flags to onboard what fritzing.org's selling at my org,but I can make a personal purchase, see if it's just bad communication or if there's intent behind it.
Regarding access to sources, frizting.org doesn't really claim the builds on sale are covered by an OSS license. They don't even mention fritzing-app by name or link to this repo.

michaellass, 2021-09-15:
The GPL license is still mentioned at https://fritzing.org/faq/ and at https://fritzing.org/support-us/ there is a link to this repository.

hoppke, 2021-09-15:
Good find. Not sure if that means what you or I would assume though, when https://fritzing.org/download/ at the same time says "The source code of Fritzing is available on our GitHub repository. Everyone is welcome to participate in the development."

closed 2021-09-30 by KjellMorgenstern