fritzing/fritzing-app #3884: Crash when right-clicking Note

daniellm19, 2021-09-25:
## Current Behaviour
Right clicking the text section of the "Note" core part in the "Tools" section when you have dragged it on to the breadboard causes the application to crash. I can't really screenshot it since the application crashes immediately.

Version 0.9.9 (bCD-348-0-f0af53a9 2021-09-22) [Qt 5.15. (the rest doesn't show up)
[image]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Step 1: Drag the "Note" core part in the "Tools" section to the breadboard.
- Step 2: Right-click the text section of the "Note" core part in the breadboard.

## Expected Behaviour
To get possible options? I was just trying to right-click and it just crashed so I can't know the expected behaviour.

daniellm19, 2021-09-25:
I just realized that someone had already written this issue, but maybe that just speeds up the fixing process.

KjellMorgenstern, 2021-09-28:
Closing as duplicate of #3880

closed 2021-09-28 by KjellMorgenstern