fritzing/fritzing-app #3885: Error while trying to instal 0.9.0

topara2005, 2021-09-30:
## Current Behaviour
Error while trying to install fritzing in windows 10
fritzing-0.9.9-l348-f0af53a9-win64.msi

[image]
[image]

**Build:**
<fritzing-0.9.9-l348-f0af53a9-win64) 

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Double click on MSI
- Step 2

## Expected Behaviour
To get the program installed

KjellMorgenstern, 2021-09-30:
Could you please copy the exact version number of you operating system?

After double clicking the MSI, the following screen should appear:
[image]
What do you mean with "Step 2" ?

topara2005, 2021-09-30:
> Could you please copy the exact version number of your operating system?
> 
[image]

> After double clicking the MSI, the following screen should appear: [image] What do you mean with "Step 2" ?
Sorry..A mistake I left it y mistake. It does not mean anything

KjellMorgenstern, 2021-09-30:
I guess the installer bails out directly after clicking the MSI file, without further interaction. 
The rendering of the font and the window are wrong.
The Windows version seems up-to-date.

A very likely cause could be a virus scanner (we tried to reproduce it with Avast today, but install worked flawless). In the past there were reports where the virus scanner blocked file access during its scan, and so the Fritzing installer got blocked from its own files because of that.

The release is quite fresh, so the virus scanners might take some more days (until they have learned the signature of our MSI and rolled out updates for their database)

Another thing to check would be our own signature. If you right-click the file and go to properties, there should be a certificate section. The MSI file is certified to "Fritzing GmbH". Probably everything is alright, otherwise you would get a warning by Windows Defender. So this is unlikely the issue, but makes sense to check to be sure (e.g. maybe Defender is disabled *and* your download was tweaked. Unlikely, but we have seen binaries modified depending on the internet provider used for downloading.

Most likely guess is currently the virus scanner. Do you have any, if yes, which one?

topara2005, 2021-09-30:
Windows defender only

vanepp, 2021-10-03:
This may be missing C++ runtime DLLs as described in this forum post although the error message doesn't sound exactly the same:

https://forum.fritzing.org/t/installation-problem-win10-64bit-visualc/13917/

checking for and/or installing the runtime DLLs is where I would start although as noted they used to be installed automatically in Win 10, it is possible that 0.9.9 is using a later version of the runtime DLLs (although 0.9.9 installed on my Win10 system without problem.)

KjellMorgenstern, 2021-10-03:
@vanepp We include the installer from Microsoft  https://aka.ms/vs/16/release/vc_redist.x64.exe . The DLLs itself are not included. As the name indicates, the installer is "redistributable", it is of course legal to include. This is the recommended way from Microsoft to resolve the MSVC DLL dependencies. The Microsoft installer will only run if the DLLs are missing or outdated. For this, it will need admin permissions. Users will be shown clear information that the DLLs need to be installed, and that Fritzing might not run if the permission is not granted. 
However, in the case of topara2005 , he said that our installer immediately fails with a missing DLL, without further dialogs or warnings. He doesn't even have a virus scanner or the like which could block anything. Also, he has Windows definer (active?), so our Fritzing GmbH certificate must be still intact, otherwise Defender would complain. So the DLL missing must be one required by the Installer itself, not by Fritzing. 

@topara2005 Could you start the installer with logging enabled? This could reduce the amount of guesswork :-)
`msiexec /i "C:\MyPackage\Example.msi" /L*V "C:\log\example.log"`