fritzing/fritzing-app #3899: Qt::SplitBehaviorFlags has not been declated error when compiling from source.

AndyAnon14, 2021-11-01:
## Current Behaviour
When compiling fritzing-app from source code on Ubuntu 20.04 with QT5 from Ubuntu repository, the following error is flagged:

/home/user/Development/fritzing_dev/fritzing-app/src/fapplication.cpp:154: error: ‘Qt::SplitBehaviorFlags’ has not been declared
  154 |     QStringList tokens = header.split(QRegExp("[ \r\n][ \r\n]*"), Qt::SplitBehaviorFlags::SkipEmptyParts);

/home/user/Development/fritzing_dev/fritzing-app/src/fapplication.cpp:170: error: ‘Qt::SplitBehaviorFlags’ has not been declared
  170 |     QStringList params = tokens.at(1).split("/", Qt::SplitBehaviorFlags::SkipEmptyParts);

**Build:**
Version 0.9.9 [Qt 5.12.8]

**Operating System:**
Ubuntu 20.04.3 LTS

**Steps to reproduce:**
- Step 1 - Installed Qt SDK on Ubuntu 20.04 from Ubuntu repositories.
- Step 2 - git cloned latest fritzing-app and fritzing-parts gits.
- Step 3 - Opened up phoenix.pro project from fritzing-app in Qt Creator and started the build.

## Expected Behaviour
Compile error free as I am using Qt 5.12 or higher as recommended.

GRS26, 2021-11-23:
did you have any luck solving this?

AndyAnon14, 2021-12-02:
No, I have not had any luck in solving the issue. I will re-attempt to compile it using a VM with a fresh install of Ubuntu. I have tried multiple ways; from compiling QTSDK from source with the recommended version and then attempting to compile Fritzing again and I still get the same results. It seems that this is completely busted on linux, which is a shame.

Anonymous-Ol, 2021-12-11:
I am having the same problem when attempting to compile fritzing on mac.

geirpa, 2021-12-17:
I found a way to solve it. Managed to build and run Fritzing.

I am running operation system: Ubuntu 20.04.3 LTS
Using Qt version 5.12.8

1) Some of these I probably had installed from before
`sudo apt-get install build-essential git cmake libssl-dev libudev-dev qt5-default libqt5serialport5-dev libqt5svg5-dev`
[https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes](url)

2) Installed libgit2
`sudo apt install libgit2-dev`
[https://stackoverflow.com/questions/61994547/how-to-fritzing-on-linux](url)

3) Installed boost
`sudo apt-get install libboost-all-dev`
[https://newbedev.com/how-to-install-boost-on-ubuntu](url)

4) Now I am trying to compile the code with, but encounter problems
`cd /path/to/fritzing-app`
 `qmake`
 `make`
[https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes](url)

5) To use the dynamic libgit2 i changed some lines in file phoenix.pro
  i) changed line 178 "LIBGIT_STATIC = true" to "LIBGIT_STATIC = false"
  ii) i got som error about not detecting boost when compiling, so i commented out line 181 in phoenix.pro `include(pri/boostdetect.pri)`
to
`#include(pri/boostdetect.pri)`
[https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing#libgit2](url)

with these changes qmake appears to work. Onwards to make:

6) Now i got to the point where i got the same error as you guys. The problem is that the fritzing code use, and thus depends on some of the newest features in Qt.
[https://forum.qt.io/topic/132252/problem-while-trying-to-build-fritzing/2](url)

Qt::SplitbehaviourFlags::SkipEmptyParts is an enum that was introduced in Qt 5.14
[https://doc.qt.io/qt-5/qt.html#SplitBehaviorFlags-enum](url)

So I thought to myself. There are 2 lines of code that doesn't work, how can i rewrite these 2 lines, to something that does the same, but can be compiled together with my version of Qt (5.12.8). I changed fapplication.cpp, line 154
`QStringList tokens = header.split(QRegExp("[ \r\n][ \r\n]*"), Qt::SplitBehaviorFlags::SkipEmptyParts);`
to
`QStringList tokens = header.split(QRegExp("[ \r\n][ \r\n]*"));`
`tokens.removeAll(QString(""));`

and fapplication.cpp, line 170
`QStringList params = tokens.at(1).split("/", Qt::SplitBehaviorFlags::SkipEmptyParts);`
to
`QStringList params = tokens.at(1).split("/");`
`params.removeAll(QString(""));`

I believe this gives the same result.
[https://doc.qt.io/qt-5/qstring.html#split](url)
[https://stackoverflow.com/questions/49872399/how-to-remove-empty-and-null-strings-from-qstringlist](url)

7) make chugged along some more, but I got some more errors. Complaining about Qt::endl and Qt::flush (These are introduced in Qt version 5.14 as well). In the file /fritzing-app/src/dialogs/fabupploadprogress.cpp. I changed Qt::endl and Qt::flush to '\n'. Seems to be the same.
[https://doc.qt.io/qt-5/qt.html#endl](url)
[https://doc.qt.io/qt-5/qt.html#flush](url)

I also got errors in /fritzing-app/src/dialogs/networkhelper.cpp. Something about qDebug() and Qt::endl. I tried to do the same here, change Qt::endl to '\n', but this time it didnt work. So I just commented out these qDebug() lines. I believe the error here was partly due to a missing include (`#include <QtDebug>`) ?

8) Anyways, after commenting out the buggy lines, make managed to compile and I got a runfile/executable file named "Fritzing" which I can run with ./Fritzing, and it starts and run the program.

Endnotes: I am just some dude with some basic knowledge in linux and in some programming languages, so if this doesn't work, or you get other errors, I cannot help you.

marcohald, 2022-02-27:
I had the same error, my Problem was that I had not set the correct PATH Variable for the QT Installation which was in my case
```export PATH=/opt/Qt/5.15.2/gcc_64/bin:$PATH```
I also needed to uninstall duplicated QT Libs with
```sudo apt remove "qt5*"```

theofficialgman, 2022-04-09:
@KjellMorgenstern can we have some clarification. was it your intention to bump the minimum requirement for fritzing to QT 5.14 up from 5.9 in this massive commit of changes https://github.com/fritzing/fritzing-app/commit/0e398a6202751bc7c6c4b2386192ec594c14dd23?

some of the comments in this thread already show you how it can be done in older QT versions, which I think is desired to avoid bloat due to QT libraries you would now have to include in the linux appimage

ErikLethDanielsen, 2022-11-30:
i had the same problem when building https://github.com/mich-w/QtSerialMonitor

error: "Obsolete Members for QString" Remove Qt::SplitBehaviorFlags::SkipEmptyParts
https://doc.qt.io/qt-5/qstring-obsolete.html 

example:
inputString.split(QRegExp("[\\n+\\r+]"), Qt::SplitBehaviorFlags::SkipEmptyParts); //Error
inputString.split(QRegExp("[\\n+\\r+]"));  //This worked

KjellMorgenstern, 2023-07-24:
We follow the frameworks policy: https://en.wikipedia.org/wiki/Qt_version_history

closed 2023-07-24 by KjellMorgenstern

mychinacoder, 2024-04-18:
> 我在构建 https://github.com/mich-w/QtSerialMonitor 时遇到了同样的问题
> 
> 错误：“QString的过时成员” 删除 Qt：：SplitBehaviorFlags：：SkipEmptyParts https://doc.qt.io/qt-5/qstring-obsolete.html
> 
> 示例：inputString.split（QRegExp（“[\n+\r+]”）， Qt：：SplitBehaviorFlags：：SkipEmptyParts）;错误 inputString.split（QRegExp（“[\n+\r+]”））;这奏效了

you are right