fritzing/fritzing-app #3901: Autorouter causes DRC errors due to traces too close.

vanepp, 2021-11-05:
## Current Behaviour

Starting from this sketch with all traces and vias in pcb deleted (remove the trailing .zip to get the .fzz file)

[Unterstand_post-neu-no-traces-no-vias.fzz.zip](https://github.com/fritzing/fritzing-app/files/7487671/Unterstand_post-neu-no-traces-no-vias.fzz.zip)

[image]

with Autorouter setting set to professional

[image]

and grid size set to the default 0.1in

[image]

run autorouter

[image]

which successfully routes the board. Now run DRC 

[image]

which reports many violations. The one circled in red on the image is too close to the entire set of pins in the part. Autorouter should be creating sufficient clearance to pass DRC. 

**Build:**

Version 0.9.9 
(bCD-348-0-f0af53a9 2021-09-22) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**

See current behaviour

## Expected Behaviour

Autorouter should produce a pcb layout that passes DRC.

KjellMorgenstern, 2021-11-13:
Can you confirm that the grid size matters?

vanepp, 2021-11-14:
No grid size doesn't appear to matter. It was initially set at 0.02in, I changed it back to the 0.1in default and redid the autoroute but DRC still failed. What does work is to increase the custom keepout value from 0.01in to 0.02in before doing the autoroute and then change it back to 0.01in to do the DRC. DRC then passes. I think this indicates we need to increase the keepout value a bit when doing the autoroute. 

[image]

but there may be a better way to achieve the same result.