fritzing/fritzing-app #3902: PCB positioning in Inspector is inaccurate

vanepp, 2021-11-05:
## Current Behaviour

Someone is trying to set holes in pcb view at exact coordinates but can not do so. First make sure snap to grid is disabled, so grid snapping isn't the problem:

[image]

place 4 holes in pcb and try and set the top left one's X coordinate to 50mm via Inspector:

[image]

when the change takes place, the x coordinate is wrong:

[image]

instead of 50mm we get 49.972mm. When I do the same to the bottom left hole (set the x coord to 50mm) we get 49.985mm (same input, two different outputs!)

[image]

entering 129 (50mm + 79mm) and 10mm on the right top hole results in 128.987mm and 10.016mm.
It would be desirable (but perhaps not possible due to floating point roundoff) to have the numbers be exact. As well it would be desirable that incrementing the value via the up and down arrows in Inspector moved in even increments (such as 1mm or .1in) which they don't at present:

[image]

1 click of the up arrow went from 128.987 to 130.009 instead of an even increment (although this is less important if exact values can be manually set!) 

[image]

**Build:**

Version 0.9.9 
(bCD-348-0-f0af53a9 2021-09-22) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**

See behaviour

## Expected Behaviour

The exact value set in to Inspector becomes the coordinate

failiz, 2021-11-15:
Duplicate of  #3639.

failiz, 2022-05-13:
@vanepp or @KjellMorgenstern , if you think that this is a duplicate of #3639, please, close the issue. If not, please, highlight the differences (or me, they are identical).

vanepp, 2022-05-29:
Yes it appears to be a dup of #3639. I'll close this one.

closed 2022-05-29 by vanepp