fritzing/fritzing-app #3903: Log while importing Bin

spassignat, 2021-11-06:
## Problem
While importing a bin part, when I have some bugs in the components, the import crashes silently (or I don't know where are the logs).
So I started fritzing in a console but nothing easy to understand:
```
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
Segmentation fault (core dumped)

``` 

 
## Proposed Solution
Could it be possible to have a global error catching and to print it in the console before exiting or, and print debug traces in the console or in a log file ?

KjellMorgenstern, 2021-11-06:
There are: A core file was dumped.

But even if you get a full backtrace, that is often not sufficient.
Much more important is to reproduce the issue with the most recent version.

KjellMorgenstern, 2021-11-13:
Debugging information and core files are already available. No information given to reproduce the crash -> I'll close the issue for now. Feel free to submit additional information for reopening.

closed 2021-11-13 by KjellMorgenstern