fritzing/fritzing-app #3905: Breadboard & Schematic do not stay in sync

BOTSOTS, 2021-11-21:
I'm new to Fritzing and have not done any of the tutorials so It's likely I am doing something wrong.  However, when I create a breadboard diagram then  go to the Schematic it is not complete.   Once I complete the schematic then return to the breadboard, it has been rearranged and some wiring is lost, plus there are a random collection of "ratsnest" lines.  When I correct this then return to the schematic, it is also rearranged, missing components and also has a random collection of "ratsnest" lines.  I've tried locking the components but this has no real effect.  

I am using version 0.9.9

KjellMorgenstern, 2021-11-21:
Not much we can do without screenshots or project file.

vanepp, 2021-11-24:
Your best bet is to post the sketch (the .fzz file) to the Fritzing forums at https://forum.fritzing.org/ upload is 7th icon from the left in the reply menu) where one of us will look it over and tell you what is wrong. The views update each other when changes are made in any view, so the recommendation is to complete one view (such as breadboard) then click on the resulting rats nest lines in the other views and route them. That isn't fool proof as sometimes unintended connections get made while routing but it works a lot better. It is possible that some of the parts you have used have problems as well. parts (especially ones from the net rather than core parts) are of variable quality and sometimes have problems. If you don't want to join the forum, you can rename the .fzz file for your sketch to project.fzz.zip and up load it here (github won't upload a .fzz file but will upload a .zip file and a .fzz file is already a zip file.)

failiz, 2023-02-22:
Not enough info and the reporter did not reply in more than a year. Close.