fritzing/fritzing-app #3912: Fritzing App not Scaling Corectly

Siphowin, 2021-12-13:
## Current Behaviour
Text
[image]
**Build:**
0.9.9

**Operating System:**
Windows 11

**Steps to reproduce:**
Install on lenovo Ideapad flex 5 with Win11

## Expected Behaviour
All the Icons are pixelated and the windows isn't properly scaled

KjellMorgenstern, 2021-12-18:
It works fine on Windows 11. Not enough information to reproduce the problem.

KjellMorgenstern, 2022-11-16:
Lenovo Ideapad flex 5 comes in 14 and 16 inch variant. We don't know the screen resolution and windows zoom configuration. Additionally, there are no other reports of the problem. 

Please reopen if you can reproduce the problem with Fritzing 0.9.10

closed 2022-11-16 by KjellMorgenstern

GoncaloTR, 2023-05-04:
I have the same problem with a flex 5 14 inch
the screen is 1920x1080 and Windows scaling is at 150%

one fix might be to change the scaling individually by using Windows troubleshooter

KjellMorgenstern, 2023-05-04:
@GoncaloTR So you have a very similar view as in the original screenshot? I would say, if the application is zoomed 150% then larger fonts, icons and bars are expected. Why is it a problem? Or are there problems with some elements not scaling with the rest?

GoncaloTR, 2023-05-04:
@KjellMorgenstern the problem is that i do not want the scaling inside the application as it is in Windows since text and graphics are pixilated

That being said, the fix in Fusion 360 was the troubleshooter and this panel (in the image below), which doesn´t solve the problem in this case. Removes the scaling, but the pixelation continues.

[image]

[image]