fritzing/fritzing-app #3918: Wire colors in Breadboard are often lost / reset

hoegge, 2021-12-31:
## Current Behaviour
Text
[image]

**Build:**
Version 0.9.9 
(bCD-348-0-f0af53a9 2021-09-22) Cocoa [Qt 5.15.2]

**Operating System:**
MacOS 10.15.7 Catalina

**Steps to reproduce:**
Hard to reproduce consistently but happens often.
- Make a breadboard design with different colors and banded colors
- Suddenly or after saving, closing down and reloading all colors of wires are the same and no banding (see picture)

## Expected Behaviour
That colors stay as defined.

hoegge, 2021-12-31:
The colors of all the blue wires were the same as the colors below the perf-board

KjellMorgenstern, 2024-09-30:
We made a lot of improvements, where not all properties were stored with the commands (the elements that build the undo/redo history) .
I tried to reproduce the issue, but did not observe it:
Save, close and open -> Wires colors are preserved :white_check_mark: 
Open, Edit, Revert -> Wire colors are preserved :white_check_mark: 

I used this sketch for testing:
[wire_colors.fz.zip](https://github.com/user-attachments/files/17186033/wire_colors.fz.zip)

It would be good to have a more details that I could try to reproduce the behavior to rule out it still happens.